The-OpenROAD-Project/OpenROAD #4467: Error: detail_route.tcl, 69 DRT-0073

oharboe, 2023-12-31:
### Describe the bug

1. untar 
2. Run same as below:

```
$ NUM_CORES=16 ./run-me-DigitalTop-asap7-base.sh 
OpenROAD v2.0-11602-g18cbc8d2b 
[deleted]
[INFO DRT-0076]   Complete 900 pins.
[INFO DRT-0077]   Complete 1000 pins.
[ERROR DRT-0073] No access point for tile_prci_domain/tile_reset_domain/boom_tile/frontend/bpd/clock.
[ERROR DRT-0073] No access point for tile_prci_domain/tile_reset_domain/boom_tile/core/FpPipeline/clock.
[ERROR DRT-0073] No access point for tile_prci_domain/tile_reset_domain/boom_tile/frontend/icache/auto_master_out_a_bits_address[0].
[ERROR DRT-0073] No access point for subsystem_l2_wrapper/l2/auto_ctrls_ctrl_in_a_bits_address[0].
[ERROR DRT-0073] No access point for tile_prci_domain/tile_reset_domain/boom_tile/dcache/auto_out_a_bits_address[0].
[INFO DRT-0077]   Complete 2000 pins.
[INFO DRT-0077]   Complete 3000 pins.
Error: detail_route.tcl, 69 DRT-0073
openroad> 

```

### Expected Behavior

No error

### Environment

```shell
OpenROAD v2.0-11602-g18cbc8d2b
```

### To Reproduce

See above

osamahammad21, 2024-01-01:
Isn't this issue a duplicate of that issue: https://github.com/The-OpenROAD-Project/OpenROAD/issues/4462

oharboe, 2024-01-01:
> Isn't this issue a duplicate of that issue: #4462 ?

Possibly, I would have to trust your judgment....

oharboe, 2024-01-03:
@osamahammad21 I think this might be a red herring. I'll refile a github issue with more detail if it recoccurs.

closed 2024-01-03 by oharboe