The-OpenROAD-Project/OpenROAD #4471: `repair_antennas` crash

kareefardi, 2024-01-02:
### Describe the bug

Crash during running `repair_antennas`.

### Expected Behavior

The command passing or a meaningful exit. 

### Environment

```shell
Git commit: 861b9b3822bd9f3d329a5f21e1795a1cf8ce7ce5
kernel: Linux 6.2.0-39-generic
os: Ubuntu 22.04.3 LTS (Jammy Jellyfish)
cmake version 3.25.1
-- The CXX compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-11613-g861b9b382
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-write-strings
[... 153 lines cut ...]
 0# 0x00005638C5AA2BFF in openroad
 1# 0x00007FBBBEC55A70 in /nix/store/qn3ggz5sf3hkjs2c797xf7nan3amdxmp-glibc-2.38-27/lib/libc.so.6
 2# odb::dbBox::getBox() in openroad
 3# odb::dbWirePathItr::getNextShape(odb::dbWirePathShape&) in openroad
 4# odb::tmg_conn::loadWire(odb::dbWire*) in openroad
 5# odb::tmg_conn::analyzeNet(odb::dbNet*) in openroad
 6# odb::orderWires(utl::Logger*, odb::dbNet*) in openroad
 7# grt::RepairAntennas::makeNetWire(odb::dbNet*, std::vector<grt::GSegment, std::allocator<grt::GSegment> >&, std::map<int, odb::dbTechVia*, std::less<int>, std::allocator<std::pair<int const, odb::dbTechVia*> > >&) in openroad
 8# grt::RepairAntennas::makeNetWires(std::map<odb::dbNet*, std::vector<grt::GSegment, std::allocator<grt::GSegment> >, grt::cmpById, std::allocator<std::pair<odb::dbNet* const, std::vector<grt::GSegment, std::allocator<grt::GSegment> > > > >&, int) in openroad
 9# grt::RepairAntennas::checkAntennaViolations(std::map<odb::dbNet*, std::vector<grt::GSegment, std::allocator<grt::GSegment> >, grt::cmpById, std::allocator<std::pair<odb::dbNet* const, std::vector<grt::GSegment, std::allocator<grt::GSegment> > > > >&, int, odb::dbMTerm*, float) in openroad
10# grt::GlobalRouter::repairAntennas(odb::dbMTerm*, int, float) in openroad
11# 0x00005638C608F1D0 in openroad
12# TclNRRunCallbacks in /nix/store/2wmwl9g0204202qpj7p53w56rmi76m92-tcl-8.6.13/lib/libtcl.so
13# 0x00007FBBC2CCD228 in /nix/store/2wmwl9g0204202qpj7p53w56rmi76m92-tcl-8.6.13/lib/libtcl.so
14# Tcl_EvalEx in /nix/store/2wmwl9g0204202qpj7p53w56rmi76m92-tcl-8.6.13/lib/libtcl.so
15# Tcl_Eval in /nix/store/2wmwl9g0204202qpj7p53w56rmi76m92-tcl-8.6.13/lib/libtcl.so
16# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in openroad
17# ord::tclAppInit(Tcl_Interp*) in openroad
18# Tcl_MainEx in /nix/store/2wmwl9g0204202qpj7p53w56rmi76m92-tcl-8.6.13/lib/libtcl.so
19# main in openroad
20# 0x00007FBBBEC3FFCE in /nix/store/qn3ggz5sf3hkjs2c797xf7nan3amdxmp-glibc-2.38-27/lib/libc.so.6
21# __libc_start_main in /nix/store/qn3ggz5sf3hkjs2c797xf7nan3amdxmp-glibc-2.38-27/lib/libc.so.6
22# _start in openroad
run.sh: line 97: 2647081 Segmentation fault      (core dumped) openroad -exit -no_splash -metrics ./or_metrics_out.json ./scripts/openroad/antenna_repair.tcl
```

eder-matheus, 2024-01-02:
@kareefardi I've tried it with the latest master branch, and the error is not happening anymore. Today we have merged a fix in GRT for pins not being properly covered, and my guess is that this crash was happening because of it.

Could you try with the latest master branch and see if it is fixed for you too?

kareefardi, 2024-01-03:
@eder-matheus I am still facing a crash using commit `0497bd0dac40e847ae0c846d3f1b7a098b95d918`. 
[out.log](https://github.com/The-OpenROAD-Project/OpenROAD/files/13820873/out.log)
I attached the full log output in case its helpful.

eder-matheus, 2024-01-04:
@kareefardi I've tried it again, running with gdb, and I still can't reproduce the crash. I'm trying now with valgrind.

kareefardi, 2024-01-09:
@eder-matheus there seems to be difference in reproducibility related to underline architecture. I will let you know what I find.

maliberty, 2024-01-29:
@kareefardi any progress or should we close this?

kareefardi, 2024-01-29:
@maliberty Sorry for the late update. We were able to reproduce this on two `amd64` machines and *not* reproduce it on `aarch64`. I also tried a debug build on the `amd64` machine but I still got a crash. I am not sure yet if this is OS specific or not. @eder-matheus what kind of machine were you using when you test the crash?

kareefardi, 2024-01-29:
However, let me test the latest version and check if it is still the case.

eder-matheus, 2024-01-29:
> @eder-matheus what kind of machine were you using when you test the crash?

@kareefardi I've tested it in a MacOS with Intel and a CentOS with Intel. Unfortunately, I don't have access to an `amd64` machine.

QuantamHD, 2024-01-29:
Try running it under ASAN and UBSAN

maliberty, 2024-01-29:
or plain old valgrind

eder-matheus, 2024-02-07:
@kareefardi Could you try this branch: https://github.com/The-OpenROAD-Project/OpenROAD/pull/4635? I recently fixed a crash in one of our designs. Perhaps this fix works for your case as well.

kareefardi, 2024-02-12:
@eder-matheus Ok I will give it a try.

kareefardi, 2024-02-29:
@eder-matheus sorry for the delay. I can no longer reproduce the crash using the recent updates.

closed 2024-02-29 by kareefardi