The-OpenROAD-Project/OpenROAD #4478: Detailed routing DRC violation that can be removed

oharboe, 2024-01-03:
### Describe the bug

Toggling routing guides on/off, we can see that the routing guide is too tight. The net is forced too close to the macro.

[image]
[image]

Later, a metal spacing DRC error can't be fixed:

[image]

### Expected Behavior

Better routing guides so that detailed routing can work.

### Environment

```shell
OpenROAD v2.0-11638-g1a366db1e
```

### To Reproduce

coming up...

oharboe, 2024-01-03:
@maliberty Reproducible case where the routing guide is too close to the macro making it impossible in detailed routing to get away from the macro.

oharboe, 2024-01-03:
> As a workaround you might try "set_macro_extension 2"

Thanks! 

I am not held up by this, I just provided this test case because, as I recall @eder-matheus was interested in a reproducible case

eder-matheus, 2024-03-28:
Using the latest version of OpenROAD, the DRCs are being fixed. We had a couple of updates in GRT related to regions near macros that probably solved the issue you was facing with this design.

closed 2024-03-28 by eder-matheus