The-OpenROAD-Project/OpenROAD #4487: Crash on "Closest match"

oharboe, 2024-01-04:
### Describe the bug

Select "Closest match" and run the command:

[image]

Crash:

```
/usr/include/c++/13/bits/stl_vector.h:1125: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = std::unique_ptr<gui::TimingPathNode>; _Alloc = std::allocator<std::unique_ptr<gui::TimingPathNode> >; reference = std::unique_ptr<gui::TimingPathNode>&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.
Signal 6 received
Stack trace:
 0# 0x000055AE7EC2FE73 in openroad
 1# 0x00007FEC31C42910 in /lib/x86_64-linux-gnu/libc.so.6
 2# pthread_kill in /lib/x86_64-linux-gnu/libc.so.6
 3# raise in /lib/x86_64-linux-gnu/libc.so.6
 4# abort in /lib/x86_64-linux-gnu/libc.so.6
 5# 0x00007FEC320E219F in /lib/x86_64-linux-gnu/libstdc++.so.6
 6# gui::TimingWidget::writePathReportCommand(QModelIndex const&, gui::TimingWidget::CommandType const&) in openroad
 7# 0x000055AE7F47D01A in openroad
 8# 0x00007FEC32F06312 in /lib/x86_64-linux-gnu/libQt5Core.so.5
 9# QAction::triggered(bool) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
10# QAction::activate(QAction::ActionEvent) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
11# 0x00007FEC33CF0D22 in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
12# 0x00007FEC33CF8BDB in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
13# QWidget::event(QEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
14# QApplicationPrivate::notify_helper(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
15# QApplication::notify(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
16# QCoreApplication::notifyInternal2(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Core.so.5
17# QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
18# 0x00007FEC33BC886A in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
19# 0x00007FEC33BCB12F in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
20# QApplicationPrivate::notify_helper(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
21# QCoreApplication::notifyInternal2(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Core.so.5
22# QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) in /lib/x86_64-linux-gnu/libQt5Gui.so.5
23# QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) in /lib/x86_64-linux-gnu/libQt5Gui.so.5
24# 0x00007FEC2E6F6F7E in /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
25# 0x00007FEC31713B2C in /lib/x86_64-linux-gnu/libglib-2.0.so.0
26# 0x00007FEC3176F46F in /lib/x86_64-linux-gnu/libglib-2.0.so.0
27# g_main_context_iteration in /lib/x86_64-linux-gnu/libglib-2.0.so.0
28# QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) in /lib/x86_64-linux-gnu/libQt5Core.so.5
29# QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) in /lib/x86_64-linux-gnu/libQt5Core.so.5
30# QCoreApplication::exec() in /lib/x86_64-linux-gnu/libQt5Core.so.5
31# gui::startGui(int&, char**, Tcl_Interp*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool) in openroad
32# ord::tclAppInit(Tcl_Interp*) in openroad
33# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
34# main in openroad
35# 0x00007FEC31C280D0 in /lib/x86_64-linux-gnu/libc.so.6
36# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
37# _start in openroad
./run-me-DigitalTop-asap7-base.sh: line 7: 3438451 Aborted                 (core dumped) openroad -gui -no_init ${SCRIPTS_DIR}/gui.tcl
```

### Expected Behavior

No crash

### Environment

```shell
OpenROAD v2.0-11638-g1a366db1e
```

### To Reproduce

1. untar https://drive.google.com/file/d/1qUp6BY_e_7MCgUG6zWe9CcxAZFDDChlc/view?usp=sharing
2. `./run-me-DigitalTop-asap7-base.sh`
3. Update timing
4. Select "Closest match" for line in snapshot above, run it
5. crash

AcKoucher, 2024-01-11:
The path that generates the crash has only clock points. Would it make sense for us to be able to write a report command for this case?

The implementation was originally thought so that the start point will always be the first non-clock point in the path.

maliberty, 2024-01-11:
What is the path?  Can you give the text version?

maliberty, 2024-01-11:
We should be able to show any path that sta can report

AcKoucher, 2024-01-11:
Printing the all nodes' names in the path I get:
```
auto_prci_ctrl_domain_reset_setter_clock_in_member_allClocks_uncore_clock
prci_ctrl_domain/_1019_/A
prci_ctrl_domain/_1019_/Y
subsystem_cbus/_3171_/A
subsystem_cbus/_3171_/Y
output131/A
output131/Y
auto_subsystem_cbus_fixedClockNode_out_clock
```

maliberty, 2024-01-11:
sta reports:
```
>>> report_checks -through auto_subsystem_cbus_fixedClockNode_out_clock
Startpoint: auto_prci_ctrl_domain_reset_setter_clock_in_member_allClocks_uncore_clock
            (input port clocked by clock_vir)
Endpoint: auto_subsystem_cbus_fixedClockNode_out_clock
          (output port clocked by clock_vir)
Path Group: clock_vir
Path Type: max

  Delay    Time   Description
---------------------------------------------------------
   0.00    0.00   clock clock_vir (rise edge)
3300.00 3300.00   clock network delay (ideal)
1700.00 5000.00 ^ input external delay
   0.00 5000.00 ^ auto_prci_ctrl_domain_reset_setter_clock_in_member_allClocks_uncore_clock (in)
53583.79 58583.79 ^ prci_ctrl_domain/_1019_/Y (BUFx2_ASAP7_75t_R)
5026.34 63610.13 ^ subsystem_cbus/_3171_/Y (BUFx2_ASAP7_75t_R)
 110.73 63720.86 ^ output131/Y (BUFx2_ASAP7_75t_R)
   0.00 63720.86 ^ auto_subsystem_cbus_fixedClockNode_out_clock (out)
        63720.86   data arrival time

8500.00 8500.00   clock clock_vir (rise edge)
3300.00 11800.00   clock network delay (ideal)
 -10.00 11790.00   clock uncertainty
   0.00 11790.00   clock reconvergence pessimism
-1700.00 10090.00   output external delay
        10090.00   data required time
---------------------------------------------------------
        10090.00   data required time
        -63720.86   data arrival time
---------------------------------------------------------
        -53630.86   slack (VIOLATED)
```
So both start & end are clocks.  I think the simplest thing is to say that if the end point is a clock then allow the start point to be one too.

PR #4528 (2024-01-15): gui: fix crash in CLOSEST_MATCH for paths with only clock nodes
Fix #4487 

Write clock nodes in situations where the path is made only of clock nodes.

Paths with only clock nodes will be written with `-through` for all nodes. Using `-from` for the first and `-to` would lead to not finding the path in report_checks.
closed 2024-01-15 by eder-matheus