The-OpenROAD-Project/OpenROAD #4489: "Closest match" doesn't show the same path as in the GUI, "Start End" does

oharboe, 2024-01-04:
### Describe the bug

Select "Closest match" and "Start End" and match them up with the GUI.

"Start End" works, "Closest match" is a very different path than the one I selected.

[image]

### Expected Behavior

Same path displayed in "Closest match"

### Environment

```shell
OpenROAD v2.0-11637-gbd1d4e1c2
```

### To Reproduce

1. untar https://drive.google.com/file/d/1n3qD9X8mX-3cAC7KjyqXa71INdnMRXhB/view?usp=sharing
2. ` ./run-me-DigitalTop-asap7-base.sh`
3. Follow instructions above

### Relevant log output

```shell
OpenROAD v2.0-11637-gbd1d4e1c2 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[WARNING STA-0357] virtual clock clock_vir can not be propagated.
Estimating parasitics
Startpoint: tile_prci_domain/tile_reset_domain/boom_tile/lsu/_278411_
            (rising edge-triggered flip-flop clocked by auto_prci_ctrl_domain_reset_setter_clock_in_member_allClocks_uncore_clock)
Endpoint: tile_prci_domain/tile_reset_domain/boom_tile/core/rename_stage/maptable/_75733_
          (rising edge-triggered flip-flop clocked by auto_prci_ctrl_domain_reset_setter_clock_in_member_allClocks_uncore_clock)
Path Group: auto_prci_ctrl_domain_reset_setter_clock_in_member_allClocks_uncore_clock
Path Type: max

  Delay    Time   Description
---------------------------------------------------------
   0.00    0.00   clock auto_prci_ctrl_domain_reset_setter_clock_in_member_allClocks_uncore_clock (rise edge)
1714.53 1714.53   clock network delay (propagated)
   0.00 1714.53 ^ tile_prci_domain/tile_reset_domain/boom_tile/lsu/_278411_/CLK (DFFHQNx2_ASAP7_75t_R)
  71.82 1786.35 v tile_prci_domain/tile_reset_domain/boom_tile/lsu/_278411_/QN (DFFHQNx2_ASAP7_75t_R)
  36.45 1822.80 ^ tile_prci_domain/tile_reset_domain/boom_tile/lsu/_200476_/Y (INVx13_ASAP7_75t_R)
[deleted]

Startpoint: tile_prci_domain/tile_reset_domain/boom_tile/lsu/_278411_
            (rising edge-triggered flip-flop clocked by auto_prci_ctrl_domain_reset_setter_clock_in_member_allClocks_uncore_clock)
Endpoint: tile_prci_domain/tile_reset_domain/boom_tile/core/rename_stage/maptable/_75733_
          (rising edge-triggered flip-flop clocked by auto_prci_ctrl_domain_reset_setter_clock_in_member_allClocks_uncore_clock)
Path Group: auto_prci_ctrl_domain_reset_setter_clock_in_member_allClocks_uncore_clock
Path Type: max

Fanout     Cap    Slew   Delay    Time   Description
-----------------------------------------------------------------------------
                          0.00    0.00   clock auto_prci_ctrl_domain_reset_setter_clock_in_member_allClocks_uncore_clock (rise edge)
                          0.00    0.00   clock source latency
     4   15.93    0.00    0.00    0.00 ^ auto_prci_ctrl_domain_reset_setter_clock_in_member_allClocks_uncore_clock (in)
                                         auto_prci_ctrl_domain_reset_setter_clock_in_member_allClocks_uncore_clock (net)
                 36.53   11.52   11.52 ^ prci_ctrl_domain/_1019_/A (BUFx2_ASAP7_75t_R)
    16   18.90   63.93   42.03   53.56 ^ prci_ctrl_domain/_1019_/Y (BUFx2_ASAP7_75t_R)
                                         net129 (net)
                 63.94    0.25   53.81 ^ load_slew2334/A (BUFx16f_ASAP7_75t_R)
    25   28.47   20.76   28.99   82.80 ^ load_slew2334/Y (BUFx16f_ASAP7_75t_R)
                                         net65571 (net)
[deleted]
```

AcKoucher, 2024-01-12:
@oharboe I just tested the following:

```
report_checks  -fall_from tile_prci_domain/tile_reset_domain/boom_tile/lsu/_278411_/QN
 -fall_through tile_prci_domain/tile_reset_domain/boom_tile/lsu/_200476_/A 
 -rise_through tile_prci_domain/tile_reset_domain/boom_tile/lsu/_200476_/Y
 ...
 -rise_to tile_prci_domain/tile_reset_domain/boom_tile/core/rename_stage/maptable/_75733_/D
 -path_delay max -fields {capacitance slew input_pins nets fanout} -format full_clock_expanded
```

Which is the result of CLOSEST_MATCH and I got the corresponding path:

```
Fanout     Cap    Slew   Delay    Time   Description
-----------------------------------------------------------------------------
                          0.00    0.00   clock auto_prci_ctrl_domain_reset_setter_clock_in_member_allClocks_uncore_clock (rise edge)
                          0.00    0.00   clock source latency

 [...clock nodes...]

     4   21.19   78.29   71.82 1786.35 v tile_prci_domain/tile_reset_domain/boom_tile/lsu/_278411_/QN (DFFHQNx2_ASAP7_75t_R)
                                         tile_prci_domain/tile_reset_domain/boom_tile/lsu/_139124_ (net)
                 95.18   18.72 1805.07 v tile_prci_domain/tile_reset_domain/boom_tile/lsu/_200476_/A (INVx13_ASAP7_75t_R)
     2   17.20   29.39   17.73 1822.80 ^ tile_prci_domain/tile_reset_domain/boom_tile/lsu/_200476_/Y (INVx13_ASAP7_75t_R)
                                         tile_prci_domain/tile_reset_domain/boom_tile/_lsu_io_core_dis_stq_idx_0[1] (net)
                 35.41    5.99 1828.80 ^ max_length46681/A (BUFx12_ASAP7_75t_R)
     3   17.50   11.48   25.13 1853.93 ^ max_length46681/Y (BUFx12_ASAP7_75t_R)
                                         net46681 (net)

[...many more nodes...]

                 49.80   13.62 8905.44 ^ load_slew527/A (BUFx16f_ASAP7_75t_R)
    47   39.83   23.90   25.18 8930.61 ^ load_slew527/Y (BUFx16f_ASAP7_75t_R)
                                         net64243 (net)
                 48.11   12.77 8943.38 ^ tile_prci_domain/tile_reset_domain/boom_tile/core/rename_stage/maptable/_75733_/CLK (DFFHQNx2_ASAP7_75t_R)
                        -10.00 8933.38   clock uncertainty
                          0.00 8933.38   clock reconvergence pessimism
                         -6.46 8926.92   library setup time
                               8926.92   data required time
-----------------------------------------------------------------------------
                               8926.92   data required time
                               -8189.77   data arrival time
-----------------------------------------------------------------------------
                                737.15   slack (MET)
```

AcKoucher, 2024-01-23:
@oharboe It seems that Closest Match is not a very different path from the one you selected.

It looks like Closest Match and From Start to End are reporting the same path, with the difference that Closest Match is expanding the clock. If you follow the nodes in the Closest Match reporting result:
```
  Delay    Time   Description
---------------------------------------------------------
   0.00    0.00   clock auto_prci_ctrl_domain_reset_setter_clock_in_member_allClocks_uncore_clock (rise edge)
1714.53 1714.53   clock network delay (propagated)
   0.00 1714.53 ^ tile_prci_domain/tile_reset_domain/boom_tile/lsu/_278411_/CLK (DFFHQNx2_ASAP7_75t_R)
  71.82 1786.35 v tile_prci_domain/tile_reset_domain/boom_tile/lsu/_278411_/QN (DFFHQNx2_ASAP7_75t_R)
  36.45 1822.80 ^ tile_prci_domain/tile_reset_domain/boom_tile/lsu/_200476_/Y (INVx13_ASAP7_75t_R)
 [...]
  ```
  You'll eventually get to the nodes:
```
[...]
                 33.76    6.13 1714.53 ^ tile_prci_domain/tile_reset_domain/boom_tile/lsu/_278411_/CLK (DFFHQNx2_ASAP7_75t_R)
     4   21.19   78.29   71.82 1786.35 v tile_prci_domain/tile_reset_domain/boom_tile/lsu/_278411_/QN (DFFHQNx2_ASAP7_75t_R)
[...]
```

Where the expanded clock ends and the rest should be the same.

oharboe, 2024-01-24:
So close?

closed 2024-01-24 by AcKoucher