The-OpenROAD-Project/OpenROAD #4511: write_macro_placement writes out empty file

oharboe, 2024-01-10:
### Describe the bug

```
$ make DESIGN_CONFIG=designs/asap7/mock-array/config.mk floorplan
[deleted]
$ make DESIGN_CONFIG=designs/asap7/mock-array/config.mk open_floorplan
[deleted]
OpenROAD v2.0-11723-gde78d0e07 
[deleted]
openroad> write_macro_placement test.tcl
openroad> 
$ cat test.tcl
$ ls -l test.tcl
$ wc -l test.tcl 
0 test.tcl
```

### Expected Behavior

Write out macro placement regardless of how the macros were placed: manually, classic mpl and mpl2.

### Environment

```shell
OpenROAD v2.0-11723-gde78d0e07
```

### To Reproduce

See above.

maliberty, 2024-01-12:
@AcKoucher you could write this in terms of odb rather than mpl2 internals.  That way it will work regardless.

AcKoucher, 2024-01-18:
@maliberty Could we do:

If mpl2 was used for macro placement, then, write_macro_placement should include only the macros placed by mpl2.
However, if another method was used, write_macro_placement should include all placed macros? In the latter we'd use odb..

?

maliberty, 2024-01-18:
I think we can replace the flag with the command and have it cover all cases.  The flag can issue a deprecation warning to use the command instead.  The command itself can live in odb as it no longer depends on the placer.

PR #4557 (2024-01-30): mpl2 & odb: move write_macro_placement command from mpl2 to odb
Fix #4511 

The idea here is that the mpl2 flow's flag `-write_macro_placement` is kept and continues to use the function `HierRTLMP::writeMacroPlacement(const std::string& file_name)`. The flag will only generate the placement of the macros that were actually placed by mpl2, differently from the command write_macro_placement, which will generate the placement of all the macros in the design.

Instead of writing the file inside C++, we return a string and write the file in .tcl.

Adding a unit test for this new utility based on a  [... cut]
closed 2024-01-30 by maliberty