The-OpenROAD-Project/OpenROAD #4512: Macro placement mirrors macros

oharboe, 2024-01-10:
Failed in global routing, ca. 7000 seconds.

`make macro_place_issue` reproducible case for mpl2: https://drive.google.com/file/d/1w_vIUNtWp7dBntP9i7WGEkgtGhTRD6M_/view?usp=sharing

I suppose I could try to flatten the design for the two macros with contention, but if macro placement can't place these two bigger macros, what are the chances it can place many smaller macros(there are SRAMs inside the branch predictor and the data cache that have all the congestion on top)?

[image]
[image]
[image]

Intentionally, BranchPredictor has all the pins on the left side. This is to break the rotational and mirroring symmetry such that mpl2 can try different directions to find one that is better.

[image]

This is odd... Why is the macro rotated R180? That puts the pins on the right hand side... I would have thought, looking at the floorplan, that having the pins on the left side was better, given the placement...

[image]

_Originally posted by @oharboe in https://github.com/The-OpenROAD-Project/OpenROAD/discussions/4504#discussioncomment-8066586_

oharboe, 2024-01-10:
If I swap the pins from one side to the other of the two macros manually by editing the placement.tcl above,  congestion is dramatically reduced.

[image]

maliberty, 2024-01-10:
@oharboe I believe you said all the pins are on one side.  That isn't true as clock is on the opposite side to the other pins (it is core facing in this placement)

@AcKoucher I think the clock pin is overweighted in the flipping decision.  The clock net has 121,095 iterms connected to it as we haven't done CTS yet.  calculateRealMacroWirelength will generate a huge value for that one net.  I think it would be better to use HPWL of the net rather than summing all the pin-pair distances.  That would de-emphasize such a large net.

maliberty, 2024-01-10:
Clock:
[image]

maliberty, 2024-01-10:
@AcKoucher I tried replacing that method with
```
float HierRTLMP::calculateRealMacroWirelength(odb::dbInst* macro)
{
  float wirelength = 0.0;

  for (odb::dbITerm* iterm : macro->getITerms()) {
    if (iterm->getSigType() != odb::dbSigType::SIGNAL) {
      continue;
    }

    odb::dbNet* net = iterm->getNet();
    if (net != nullptr) {
      const odb::Rect bbox = net->getTermBBox();
      wirelength += bbox.dx() + bbox.dy();
    }
  }

  return wirelength;
}
```
and running this block.  I now see R0 instead of R180.  Please take over from here.

closed 2024-01-12 by maliberty