The-OpenROAD-Project/OpenROAD #4513: Strange leaves on CTS tree

oharboe, 2024-01-10:
CTS takes ca. 25000 seconds on my machine...

To reproduce:

1. untar https://drive.google.com/file/d/13eenpP2JgzXJD0uP3KIpX4QLArBVjzT2/view?usp=sharing
2. 

Yellow are hold cells, very reasonable... Clock tree looks odd. Those leaf nodes on the right side are flip flops. It would be nice if there was some clear visual indication of macros vs. flip flop endpoints in the CTS view....

[image]

[image]

[image]

```
report_checks -path_delay max -fields {fanout skew}
Startpoint: reset (input port clocked by clock_vir)
Endpoint: system/prci_ctrl_domain/_3041_
          (recovery check against rising-edge clock fake_pll_clk)
Path Group: asynchronous
Path Type: max

Fanout   Delay    Time   Description
---------------------------------------------------------------
          0.00    0.00   clock clock_vir (rise edge)
       1000.00 1000.00   clock network delay (ideal)
       1700.00 2700.00 v input external delay
     1    0.00 2700.00 v reset (in)
     1   10.58 2710.58 v hold4662/Y (BUFx2_ASAP7_75t_R)
     1   12.61 2723.18 v hold4387/Y (BUFx2_ASAP7_75t_R)
     1   12.49 2735.67 v hold4663/Y (BUFx2_ASAP7_75t_R)
     1   12.65 2748.32 v hold4335/Y (BUFx2_ASAP7_75t_R)
     1   12.65 2760.97 v hold4664/Y (BUFx2_ASAP7_75t_R)
     1   12.72 2773.69 v hold4388/Y (BUFx2_ASAP7_75t_R)
     1   12.81 2786.50 v hold4665/Y (BUFx2_ASAP7_75t_R)
     1   12.50 2799.00 v hold64/Y (BUFx2_ASAP7_75t_R)
     1   12.66 2811.66 v hold4666/Y (BUFx2_ASAP7_75t_R)
     1   12.66 2824.31 v hold4389/Y (BUFx2_ASAP7_75t_R)
     1   12.59 2836.91 v hold4667/Y (BUFx2_ASAP7_75t_R)
     1   12.71 2849.61 v hold4336/Y (BUFx2_ASAP7_75t_R)
     1   12.48 2862.09 v hold4668/Y (BUFx2_ASAP7_75t_R)
     1   12.39 2874.48 v hold4390/Y (BUFx2_ASAP7_75t_R)
     1   12.40 2886.88 v hold4669/Y (BUFx2_ASAP7_75t_R)
     1   12.34 2899.21 v hold56/Y (BUFx2_ASAP7_75t_R)
     1   12.32 2911.54 v hold4670/Y (BUFx2_ASAP7_75t_R)
     1   12.23 2923.77 v hold4391/Y (BUFx2_ASAP7_75t_R)
     1   12.45 2936.22 v hold4671/Y (BUFx2_ASAP7_75t_R)
     1   12.49 2948.71 v hold4337/Y (BUFx2_ASAP7_75t_R)
     1   12.35 2961.06 v hold4672/Y (BUFx2_ASAP7_75t_R)
     1   12.16 2973.22 v hold4392/Y (BUFx2_ASAP7_75t_R)
     1   12.26 2985.48 v hold4673/Y (BUFx2_ASAP7_75t_R)
     1   12.38 2997.87 v hold65/Y (BUFx2_ASAP7_75t_R)
     1   12.25 3010.12 v hold4674/Y (BUFx2_ASAP7_75t_R)
     1   12.19 3022.31 v hold4393/Y (BUFx2_ASAP7_75t_R)
     1   12.21 3034.52 v hold4675/Y (BUFx2_ASAP7_75t_R)
     1   12.21 3046.72 v hold4338/Y (BUFx2_ASAP7_75t_R)
     1   12.47 3059.19 v hold4676/Y (BUFx2_ASAP7_75t_R)
     1   12.59 3071.78 v hold4394/Y (BUFx2_ASAP7_75t_R)
     1   15.63 3087.41 v hold4677/Y (BUFx2_ASAP7_75t_R)
     1   18.70 3106.12 v input5/Y (BUFx24_ASAP7_75t_R)
     1   13.56 3119.68 v hold4678/Y (BUFx2_ASAP7_75t_R)
     1   14.47 3134.15 v hold4395/Y (BUFx2_ASAP7_75t_R)
     1   15.29 3149.44 v hold4339/Y (BUFx2_ASAP7_75t_R)
     1   15.29 3164.73 v hold4396/Y (BUFx2_ASAP7_75t_R)
     1   15.27 3180.00 v hold66/Y (BUFx2_ASAP7_75t_R)
     1   15.29 3195.29 v hold4397/Y (BUFx2_ASAP7_75t_R)
     1   15.25 3210.53 v hold4340/Y (BUFx2_ASAP7_75t_R)
     1   15.21 3225.74 v hold4398/Y (BUFx2_ASAP7_75t_R)
     1   15.29 3241.03 v hold57/Y (BUFx2_ASAP7_75t_R)
[... 813 lines cut ...]
     1   25.37 2759.03 v wire96183/Y (BUFx12f_ASAP7_75t_R)
    13  259.71 3018.74 v system/serial_tl_domain/_1128_/SN (HAxp5_ASAP7_75t_R)
     3  104.06 3122.80 v system/serial_tl_domain/_0882_/Y (OR3x4_ASAP7_75t_R)
     1   13.63 3136.44 ^ system/serial_tl_domain/_0897_/Y (AOI211x1_ASAP7_75t_R)
     1   22.32 3158.76 ^ system/serial_tl_domain/_0898_/Y (AO221x1_ASAP7_75t_R)
     1   23.94 3182.70 ^ system/serial_tl_domain/_0899_/Y (OR4x2_ASAP7_75t_R)
     1   20.19 3202.89 ^ system/serial_tl_domain/_0900_/Y (OA22x2_ASAP7_75t_R)
          0.01 3202.90 ^ system/serial_tl_domain/_1190_/D (DFFHQNx2_ASAP7_75t_R)
               3202.90   data arrival time

       8500.00 8500.00   clock serial_tl_0_clock (rise edge)
         99.58 8599.58   clock network delay (propagated)
        -10.00 8589.58   clock uncertainty
          0.00 8589.58   clock reconvergence pessimism
               8589.58 ^ system/serial_tl_domain/_1190_/CLK (DFFHQNx2_ASAP7_75t_R)
         -5.33 8584.25   library setup time
               8584.25   data required time
---------------------------------------------------------------
               8584.25   data required time
               -3202.90   data arrival time
---------------------------------------------------------------
               5381.35   slack (MET)
```

_Originally posted by @oharboe in https://github.com/The-OpenROAD-Project/OpenROAD/discussions/4504_

precisionmoon, 2024-01-11:
Can you clarify what the request is?   Is this about enhancing Clock Tree Viewer to show FF leaves differently from macro cell leaves?

oharboe, 2024-01-11:
@precisionmoon I was curious as to why the clock tree looked normal, except for those leaves to the right.

However, now that I have significantly modified and I understand better the MegaBoom code, I no longer see this effect, so close this for now and I'll reopen or file a new issue if I see something similar again.

The MegaBoom code used to create this clock tree had some deficiencies w.r.t. clocks because there was a PLL in it that should not be there, so this is a strange effect is for a pooly understood Verilog...

closed 2024-01-11 by oharboe