The-OpenROAD-Project/OpenROAD #4533: Reduce running time for report

oharboe, 2024-01-15:
### Description

1. untar https://drive.google.com/file/d/1OhTilBdF_m8QiEw4ujTOXB-seuToESH_/view?usp=sharing
2. `./run-me-ChipTop-asap7-flat.sh`, which runs `report_metrics 5 "global route`

Takes a long time, longer than global routing.

### Suggested Solution

Speed up the above

### Additional Context

This is from MegaBoom.

maliberty, 2024-01-15:
@gadfort I'm guessing this will be related to the psm work you are planning to do.

gadfort, 2024-01-15:
@maliberty I don't think so. Looking at the report_metrics in ORFS there is no call to psm. Only calls to STA. Might be helpful if @oharboe is able to narrow down the offending calls inside that function.

maliberty, 2024-01-15:
I was thinking reporting meant the final report step that includes psm.  Narrowing would be helpful.

oharboe, 2024-01-15:
> I was thinking reporting meant the final report step that includes psm. Narrowing would be helpful.

Yes: all the time is spent in OpenSTA, so changing how OpenSTA is used or changing OpenSTA itself that could move the needle.

The test case is quick to run: fire it up in the debugger suspend/resume and you'll see after a few seconds and minutes where time is spent.

Some of this may be fixable by modifying `Metrics.tcl`; a single pass over the data to collect all information instead of many passes to collect pieces of information:

https://github.com/The-OpenROAD-Project/OpenROAD/blob/fcd96313df26a42bfd967a10b42c7baaa4033a07/src/Metrics.tcl#L128

maliberty, 2024-01-15:
I expect most of the time will be in recomputing the timing & power not metric collection.