The-OpenROAD-Project/OpenROAD #4539: rtlmp fails without SYNTH_HIERARCHICAL=1

oharboe, 2024-01-16:
### Describe the bug

1. untar https://drive.google.com/file/d/1aGCnX_eXryTUB5RYXcMJ59_xdw9lVIeK/view?usp=sharing
2. Run below

```
$ NUM_CORES=16 ./run-me-ChipTop-asap7-flat.sh 
OpenROAD v2.0-11843-gc74f6a2f0 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ORD-0030] Using 16 thread(s).
HierRTLMP Flow enabled...
rtl_macro_placer -halo_width 20 -report_directory bazel-bin/objects/asap7/ChipTop/flat/rtlmp
Floorplan Outline: (0.0, 0.0) (1500.0, 1500.0),  Core Outline: (2.052, 2.16) (1497.96, 1497.96)
Traversed logical hierarchy
	Number of std cell instances: 940301
	Area of std cell instances: 120349.56
	Number of macros: 76
	Area of macros: 181911.06
	Area of macros with halos: 565030.88
	Area of std cell instances + Area of macros: 302260.62
	Core area: 2237579.00
	Design Utilization: 0.14
	Core Utilization: 0.06
	Manufacturing Grid: 1
```

### Expected Behavior

No failure

### Environment

```shell
OpenROAD v2.0-11843-gc74f6a2f0
```

### To Reproduce

See above

maliberty, 2024-04-18:
@AcKoucher have you looked at this?

AcKoucher, 2024-04-18:
@maliberty 

The failure in the issue was due to the fact that the overlap penalty bug was having a big impact on soft SA cost, so the generated floorplan would never fit in the outline.

This is the final result from an early run in SA back from when the overlap penalty wasn't yet fixed (blue for std cell cluster, purple mixed, red hard):

[image]

After the fix, it started failing in the next soft SA level, because apparently is very hard for SA to fit all macros into the outline with all the different tilings:

[image]

I haven't run this after the boundary penalty fix, but I suspect is going to similarly hard to fit. However, what concerns me is that, as we can see on the second image, literally all other clusters are fixed terminals. i.e. even if we're able to fit all the macros in the outline, the routers will have a very though time as there will probably be very high congestion in between these hard macros. This makes me wonder if this wouldn't also be an issue related to clustering rather than only SA.

maliberty, 2024-04-18:
Are all the blocks hard except 20 in blue?  Is the context a mix of hard and soft?

AcKoucher, 2024-04-18:
Yes. What we see is in the second image is actually a second level, meaning we're trying to place the children clusters of the single mixed cluster "0" we see in the first image.

When I saw "flattened" my first thought was that clustering would result in a single level tree.

AcKoucher, 2024-05-03:
@oharboe @maliberty I just tested with master and MPL finishes without errors. It looks like the penalty fixes + sequence pair fixed had some meaningful impact here.

<image src="https://github.com/The-OpenROAD-Project/OpenROAD/assets/104802710/3bea2fbc-fb39-4a54-b9dc-ddf0ab791a4b" width="400">

maliberty, 2024-05-04:
@oharboe can you test the full flow with the latest?  That macro placement doesn't look great but perhaps it can go through.

maliberty, 2024-05-04:
@AcKoucher do you have a sense of why it is so compressed to one corner?

oharboe, 2024-05-04:
@maliberty @AcKoucher Ouch, that macro placement looks pretty painful because there's normally macros in all the corners and a large blob of logic in the middle. Will run and see what happens.

oharboe, 2024-05-04:
It fails in global routing:

```
Floorplan Outline: (0.0, 0.0) (1194.54, 1194.54),  Core Outline: (4.05, 4.05) (1190.538, 1190.43)
Traversed logical hierarchy
        Number of std cell instances: 923831
        Area of std cell instances: 118383.91
        Number of macros: 72
        Area of macros: 154072.59
        Area of macros with halos: 507326.91
        Area of std cell instances + Area of macros: 272456.50
        Core area: 1407625.50
        Design Utilization: 0.19
        Core Utilization: 0.09
        Manufacturing Grid: 1
```

[image]

maliberty, 2024-05-04:
Your placement doesn't appear to match @AcKoucher result.  Neither looks great though.

oharboe, 2024-05-04:
> Your placement doesn't appear to match @AcKoucher result. Neither looks great though.

I no loger have access to the exact version that produced that reproduction case, so I made a fresh from megaboom master.

AcKoucher, 2024-05-04:
@oharboe I still can work on the original issue to investigate, but the additional case is helpful.

@maliberty The problem seems to be that multilevel auto-clustering is doing a very poor job. Creating a secondary level that has all the macro clusters and only a single std cell cluster feels completely wrong.

So far, all the fixed for mpl2 were related to simulated annealing and its abstractions. This looks like a very good test case to start digging into clustering.

PR #5292 (2024-06-27): mpl2: include macros in large flat cluster partitioning
Resolve #4539

Include macros in PAR partitioning when breaking flat clusters. The original code was only splitting the std cells between the generated partitions leading to something like the image below for flat megaBoom.

[image]

With the changes here:
[image]
closed 2024-06-27 by maliberty