The-OpenROAD-Project/OpenROAD #4540: Make OpenROAD GUI responsive when it is processing commands

oharboe, 2024-01-16:
### Description

Starting the GUI takes a long time because there is a lot to load. It would be nice to be able to use the GUI while it is busy processing, queuing up more commands.

Use case:

1. Start GUI
2. Load DRC report
3. Update timing
4. Update clock tree
5. Type "report_clock_skew" in shell.

Now wait for all the results without having sit and stare until the GUI is ready to accept the next button to be pressed.

### Suggested Solution

Process in background, leave GUI responsive to write more commands on shell and/or click buttons to line up more work.

maliberty, 2024-01-19:
In general it isn't safe for the GUI to be loading from odb while something else is modifying it.  If we allow arbitrary actions it will lead to weird random problems.  We would have to define a limited subset that would be read-only operations.  Its hard to make it a general purpose solution.

maliberty, 2024-01-19:
See https://github.com/The-OpenROAD-Project/OpenROAD/issues/1957

maliberty, 2024-01-19:
I'm inclined to think this is a duplicate.  Please close if you agree or describe how it differs.

oharboe, 2024-01-19:
Duplicate of #1957

closed 2024-01-19 by oharboe