The-OpenROAD-Project/OpenROAD #4548: Global place: Command terminated by signal 11

oharboe, 2024-01-18:
### Describe the bug

```
[NesterovSolve] Iter: 400 overflow: 0.400787 HPWL: 24931330112
[NesterovSolve] Iter: 410 overflow: 0.353168 HPWL: 24745674434
[NesterovSolve] Iter: 420 overflow: 0.305717 HPWL: 24289745133
[INFO GPL-0100] worst slack -3.76e-07
[INFO GPL-0103] Weighted 174499 nets.
[NesterovSolve] Iter: 430 overflow: 0.275646 HPWL: 23622015587
[NesterovSolve] Iter: 440 overflow: 0.244673 HPWL: 22883279513
[NesterovSolve] Iter: 450 overflow: 0.218247 HPWL: 22215706593
[NesterovSolve] Iter: 460 overflow: 0.198184 HPWL: 21474367018
[INFO GPL-0100] worst slack -3.76e-07
[INFO GPL-0103] Weighted 174498 nets.
[INFO GPL-0075] Routability numCall: 1 inflationIterCnt: 1 bloatIterCnt: 0
Command terminated by signal 11
Elapsed time: 1:25:43[h:]min:sec. CPU time: user 8528.74 sys 61.06 (167%). Peak memory: 63194000KB.
make[1]: *** [/home/oyvind/OpenROAD-flow-scripts/flow/Makefile:668: do-3_3_place_gp] Error 139
```

### Expected Behavior

No crash

### Environment

```shell
OpenROAD v2.0-11865-g6471c0995
```

### To Reproduce

See above

maliberty, 2024-01-18:
@eder-matheus It might be in routability, its hard to tell from the log.  Please take a look

oharboe, 2024-01-18:
@eder-matheus I have a deltaDebug.py running, it will hopefully reduce the reproduction time from hours to minutes... Stay tuned...

```
$ openroad -python ~/OpenROAD-flow-scripts/tools/OpenROAD/etc/deltaDebug.py --persistence 3 --use_stdout --error_string "Segmentation fault" --base_db_path bazel-bin/results/asap7/ChipTop/flat/3_2_place_iop.odb --step ./run-me-ChipTop-asap7-flat.sh
OpenROAD v2.0-11865-g6471c0995
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[WARNING ORD-0039] .openroad ignored with -python
Backing up original base file.
Performing a step with the original input file to calculate timeout.
Error Code found: Segmentation fault
Step 1, Insts level debugging, Insts 1907862, Nets 1750112, cut elements 953931, timeout 108 minutes
[C#]
...
```

oharboe, 2024-01-23:
I have it running in the debugger now as well as trying to whittle down the test case. Whittling down the test case has prove very, very tough(90 minutes for one iteration and it has been running for days...).

It hasn't crashed in the debugger yet, it is running for a long time, but I thought I would attach a screenshot with a stacktrace.

```
[NesterovSolve] Iter: 430 overflow: 0.275646 HPWL: 23622015587
[NesterovSolve] Iter: 440 overflow: 0.244673 HPWL: 22883279513
[NesterovSolve] Iter: 450 overflow: 0.218247 HPWL: 22215706593
[NesterovSolve] Iter: 460 overflow: 0.198184 HPWL: 21474367018
[INFO GPL-0100] worst slack -3.76e-07
[INFO GPL-0103] Weighted 174498 nets.
[INFO GPL-0075] Routability numCall: 1 inflationIterCnt: 1 bloatIterCnt: 0
[no further output before the Segfault]
```

Suspending while it is running here, gives me this stack trace.

[image]

oharboe, 2024-01-23:
@eder-matheus 9 hours later in debugger...

What can I check here?

[image]

eder-matheus, 2024-01-23:
Something is off here. From what I see, the function mazeRouteMSMD is called, but it should not run inside routability-driven global placement, as we don't want to run the overflow iterations there. I will check the code and try running the test case you provided to see if my changes can fix it.

oharboe, 2024-01-23:
Wondering around the code... Is this right? Can't this overflow, in theory?

https://github.com/The-OpenROAD-Project/OpenROAD/blob/c1fc5f730a59874a1b9735be82d267da8417c6e7/src/grt/src/fastroute/src/utility.cpp#L67

oharboe, 2024-01-23:
@eder-matheus Thoery: the large clock tree in MegaBoom has enough pins that some static limit is overflowing. This would make it impossible to whittle down the example.

[image]

eder-matheus, 2024-01-23:
@oharboe This makes sense. I'll try to get rid of these static limits. I thought I'd removed all of them in the past, but we have a few ones hanging in the code.

oharboe, 2024-01-23:
> @oharboe This makes sense. I'll try to get rid of these static limits. I thought I'd removed all of them in the past, but we have a few ones hanging in the code.

Very good. I'm glad this issue is unstuck now at least ...

The whittling down is not going to work.

#4571 is a "success", lots of failures in fast tests...

eder-matheus, 2024-01-23:
This is a good start! Thanks for the help debugging it. I will start the work to remove the static limits today and hopefully will have the fix completed soon.

eder-matheus, 2024-01-26:
@oharboe With this PR https://github.com/The-OpenROAD-Project/OpenROAD/pull/4591, I could complete the placement stage in the test case you provided. I search the code for other large arrays and turn them all into vectors, so GRT should be safe now regarding out-of-stack issues.

oharboe, 2024-01-26:
@eder-matheus Please go ahead and merge. If there are further fixes needed, we can do those in later PRs.

closed 2024-01-26 by maliberty