The-OpenROAD-Project/OpenROAD #4570: Handle `ERROR: ABC contains local modifications! Set ABCREV=default in Yosys Makefile!` automatically in `build_openroad.sh`

oharboe, 2024-01-23:
### Description

The fix for the error below is to run `rm -rf tools/yosys/abc/`, which is not at all obvious.

```
$ ./build_openroad.sh --local --latest --openroad-args -DCMAKE_BUILD_TYPE=DEBUG
[INFO FLW-0027] Saving logs to build_openroad.log
[INFO FLW-0028] ./build_openroad.sh --local --latest --openroad-args -DCMAKE_BUILD_TYPE=DEBUG
[INFO FLW-0002] Updating git submodules.
Submodule path 'tools/OpenROAD': checked out '04d4677f63d150d30c4d346f20b2f861a77607e1'
Submodule path 'tools/OpenROAD/third-party/abc': checked out '95b3543e928640dfa25f9e882e72a090a8883a9c'
[INFO FLW-0004] Updating OpenROAD app to the HEAD  of origin/master.
Previous HEAD position was 04d4677f6 Merge pull request #4520 from eder-matheus/grt_restore_overlap_fix
HEAD is now at c1fc5f730 Merge pull request #4568 from eder-matheus/grt_ant_fix
From https://github.com/The-OpenROAD-Project/OpenROAD
 * branch                master     -> FETCH_HEAD
Already up to date.
[INFO FLW-0001] Using local build method. This will create binaries at 'tools/install' unless overwritten.
[INFO FLW-0017] Compiling Yosys.
make: Entering directory '/home/oyvind/OpenROAD-flow-scripts/tools/yosys'
[  0%] Building abc/abc-896e5e7
[ 10%] Building share/include/kernel/cost.h
ERROR: ABC contains local modifications! Set ABCREV=default in Yosys Makefile!
make: *** [Makefile:802: abc/abc-896e5e7] Error 1
make: *** Waiting for unfinished jobs....
make: Leaving directory '/home/oyvind/OpenROAD-flow-scripts/tools/yosys'
```

### Suggested Solution

Add automation to handle this circumstance in `build_openroad.sh`

maliberty, 2024-01-23:
How did you get local modifications to abc?  Its not clear what the 'handling' here is.  There is no way to know what the user's intention is (locally modified abc, or some mistake requiring rm).

oharboe, 2024-01-23:
> How did you get local modifications to abc? Its not clear what the 'handling' here is. There is no way to know what the user's intention is (locally modified abc, or some mistake requiring rm).

I don't know. I have have been building with `./build_openroad --local --latest` and this morning it didn't work.

Was yosys recently updated?

If so, then my local build folders might be out of date after the upgrade and you should expect others to run into the same issue.

maliberty, 2024-01-23:
The yosys submodule was not updated in ORFS so I don't know how it would have happened.  We did update the yosys repo but that should have no effect.  Let's see if others report the same.

oharboe, 2024-01-23:
Minimally I wanted to get the workaround written down...

vvbandeira, 2024-01-23:
@oharboe could you try using `--clean`?
I remember we had a similar issue a while back, and the solution was adding this to the `--clean`:
https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/0ea230930b57a6fb517521d8c34cb3591815b596/build_openroad.sh#L362-L366

closed 2024-01-23 by oharboe