The-OpenROAD-Project/OpenROAD #4577: Build error OpenRoad Flow Scripts

Revenantx86, 2024-01-24:
### Description

I have run the setup.sh and dependency installer but when I try to run build I get the following error. Any Idea how I can solve this issue ? 
[image]

luarss, 2024-01-24:
Hi, for ORFS local installation may I check if you followed these instructions?

```
git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts
cd OpenROAD-flow-scripts
sudo ./setup.sh
./build_openroad.sh --local
```

oharboe, 2024-01-24:
@Revenantx86 You probably have a corrupted /usr/local folder that you need to clean out.

When you follow https://openroad-flow-scripts.readthedocs.io/en/latest/user/BuildLocally.html, the files are not installed in /usr/local, but in `OpenROAD-flow-scripts/dependencies`:

```
$ find . -regex .*spdlog/logger\.h
./dependencies/include/spdlog/logger.h
```

Revenantx86, 2024-01-25:
Thank you very much I have fixed the problem by installing the dependencies properly.

closed 2024-01-25 by Revenantx86