The-OpenROAD-Project/OpenROAD #4582: Support spacing range in drt

maliberty, 2024-01-24:
### Describe the bug

There are some PDKs that still use the older range still spacing rules, eg:

```
LAYER layerName
TYPE ROUTING ;
  SPACING minSpacing
    RANGE minWidth maxWidth
```
We don't need to support the sub-options but this basic syntax is needed.  Currently this triggers:

```
    if (rule->hasRange()) {
      logger_->warn(DRT, 140, "SpacingRange unsupported.");
```

and leads to incorrect results.

### Expected Behavior

Support this rule.

### Environment

```shell
Head of OR
```

### To Reproduce

Private PDK so a manual test will have to be generated.

closed 2024-01-29 by osamahammad21

vijayank88, 2024-01-30:
`gf180` produces similar warning message.
```
detailed_route -output_drc ./reports/gf180/spm/base/5_route_drc.rpt -output_maze ./results/gf180/spm/base/maze.log -bottom_routing_layer Metal2 -top_routing_layer Metal5 -disable_via_gen -save_guide_updates -verbose 1 -drc_report_iter_step 5
[INFO DRT-0149] Reading tech and libs.
[WARNING DRT-0140] SpacingRange unsupported.
[WARNING DRT-0140] SpacingRange unsupported.
```

maliberty, 2024-01-30:
What commit id did you use when you got the gf180 result?

vijayank88, 2024-01-30:
With latest OR commit, warning not exist...