The-OpenROAD-Project/OpenROAD #4583: Undefined symbols : absl::lts_20230802::container_internal::kEmptyGroup

stefanottili, 2024-01-24:
### Describe the bug

MacOS M1 homebrew:
 I've successfully been building openroad including mpl2/par with the patch + workaround described in https://github.com/The-OpenROAD-Project/OpenROAD/pull/3643

I made the mistake to run brew update and reran the OpenRoad/etc/DependencyInstaller.sh after running into issues. In order to get to the final like command I did the following:

a) downgrade protobuf from the default @25 to @21
```
brew uninstall protobuf
brew install protobuf@21
```
b) comment out one line referencing SWIG_snprintf in src/gui/tclSwig.h
probably not the right way, but it's in an error path and get's the compile going again
1627 //    SWIG_snprintf(temp, sizeof(temp), "%d", argno+1);

c) DependencyInstaller installs or-tools 9.5.2237

I previously gave 9.8.xxxx a try, but that didn't fix the dylib issue, so 9.5.2237 should be fine, worked before.

But now I'm getting this absl link error and dont' know how to proceed, any hint welcome.

```ld: Undefined symbols:
  absl::lts_20230802::container_internal::kEmptyGroup, referenced from:
      gpl::MBFF::RunILP(std::__1::vector<gpl::Flop, std::__1::allocator<gpl::Flop>> const&, std::__1::vector<gpl::Tray, std::__1::allocator<gpl::Tray>> const&, std::__1::vector<std::__1::pair<int, int>, std::__1::allocator<std::__1::pair<int, int>>>&, double, int) in gpl.a[17](mbff.cpp.o)
      gpl::MBFF::RunILP(std::__1::vector<gpl::Flop, std::__1::allocator<gpl::Flop>> const&, std::__1::vector<gpl::Tray, std::__1::allocator<gpl::Tray>> const&, std::__1::vector<std::__1::pair<int, int>, std::__1::allocator<std::__1::pair<int, int>>>&, double, int) in gpl.a[17](mbff.cpp.o)
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

### Expected Behavior

successful linking openroad

### Environment

```shell
Git commit: d1f4cec2596dbfab3dc68f7a6564ba555cb999cd
kernel: Darwin 23.3.0
os: macOS 14.3
cmake version 3.28.1
CMake Deprecation Warning at third-party/abc/CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Warning at src/CMakeLists.txt:245 (message):
  spdlog: SPDLOG_FMT_EXTERNAL=ON

-- The CXX compiler identification is AppleClang 15.0.0.15000100
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-11992-gd1f4cec25
-- System name: Darwin
-- Compiler: AppleClang 15.0.0.15000100
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is AppleClang 15.0.0.15000100
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
[... 103 lines cut ...]
-- Found OpenMP_CXX: -Xclang -fopenmp (found version "5.0") 
-- Found OpenMP: TRUE (found version "5.0")  
-- GPU is not enabled
-- TCL library: /opt/homebrew/opt/tcl-tk/lib/libtcl8.6.dylib
-- TCL header: TCL_HEADER-NOTFOUND
-- Found Eigen3: /opt/homebrew/opt/or-tools/share/eigen3/cmake/Eigen3Config.cmake (found version "3.4.0") 
-- GUI is enabled
-- Charts widget is enabled
-- Found Boost: /opt/homebrew/lib/cmake/Boost-1.83.0/BoostConfig.cmake (found version "1.83.0") found components: serialization 
-- Could NOT find VTune (missing: VTune_LIBRARIES VTune_INCLUDE_DIRS) 
-- Found Boost: /opt/homebrew/lib/cmake/Boost-1.83.0/BoostConfig.cmake (found suitable version "1.83.0", minimum required is "1.78")  
-- TCL library: /opt/homebrew/opt/tcl-tk/lib/libtcl8.6.dylib
-- TCL header: TCL_HEADER-NOTFOUND
-- Found Boost: /opt/homebrew/lib/cmake/Boost-1.83.0/BoostConfig.cmake (found version "1.83.0") found components: serialization system thread 
-- TCL readline disabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done (6.0s)
-- Generating done (0.7s)
```

### To Reproduce

cd OpenRoad/tools/build
make

maliberty, 2024-01-24:
I believe absl is used from or_tools so perhaps it is the usual problem there on mac?

stefanottili, 2024-01-24:
yes, absl is part of the ortools.

But this is an issue I've not seen before, I've always been able to link openroad.

The "avoid the runtime error" workaround is to put the ortools..dylib at the end of the link command, but I don't get that far as of today.

maliberty, 2024-01-24:
gpl::MBFF::RunILP also calls into or-tools now.  Perhaps your workaround needs a tweak.  Though that change is not super recent.  When did you last update successfully?

stefanottili, 2024-01-25:
I compiled yesterday’s master before a “brew upgrade” broke something and I ran the DependencyInstaller again.

maliberty, 2024-01-25:
Perhaps your version of absl is out of sync with what or-tools requires?  Or perhaps you two versions of absl installed?

Have you tried a clean build?

stefanottili, 2024-01-25:
I’ve tried both formerly working or-tools version and also completely wiped out OpenRoad, cloned it again and ran the DependencyInstaller.sh. What worked yesterday doesn’t work today. I’ve also updated macOS, that might have hanged the linker …

stefanottili, 2024-01-25:
nope, ld -v 'BUILD 13:21:42 Nov 10 2023' shows that it didn't change lately.

I'm at a loss here. Any other arm64 MacOS OpenRoad users out there ?

maliberty, 2024-01-25:
@eder-matheus are you on arm64?

eder-matheus, 2024-01-25:
> @eder-matheus are you on arm64?

No, I have an intel cpu.

stefanottili, 2024-01-25:
on macOS Sonoma the etc/DependencyInstaller.sh or-tools 9.5.2237 causes this issue.

Either install or-tools 9.8 stand alone or patch the etc/DependencyInstaller.sh to be able to link the openroad executable.
```
208,209c208,209
<     orToolsVersionBig=9.8
<     orToolsVersionSmall=${orToolsVersionBig}.3296
---
>     orToolsVersionBig=9.5
>     orToolsVersionSmall=${orToolsVersionBig}.2237
712c712
<         _installOrTools "macOS" "14.1" $(uname -m)
---
>         _installOrTools "macOS" "13.0.1" $(uname -m)
```
With the dylib workaround I get an executable without runtime error. 

I used to do "make install", but that fails today with 
```
Install the project...
-- Install configuration: "RELEASE"
-- Installing: /usr/local/bin/sta
CMake Error at src/sta/cmake_install.cmake:41 (file):
  file INSTALL cannot copy file
  "/Users/stefanthiede/OpenROAD-flow-scripts/tools/OpenROAD/src/sta/app/sta"
  to "/usr/local/bin/sta": Permission denied.
Call Stack (most recent call first):
  src/cmake_install.cmake:57 (include)
  cmake_install.cmake:47 (include)
```

maliberty, 2024-01-25:
@vvbandeira please look at updating or-tools as above

eder-matheus, 2024-04-20:
Fixed in https://github.com/The-OpenROAD-Project/OpenROAD/pull/4978

closed 2024-04-20 by eder-matheus