The-OpenROAD-Project/OpenROAD #4594: Surprisingly large clock rise time 332000ps, expected 500ps

oharboe, 2024-01-26:
### Describe the bug

1. untar https://drive.google.com/file/d/19nK2MBVsMEgIc7n-mrFOWnPK6t5RBTnx/view?usp=sharing
2. Run `./gui.sh`
3. Enter `report_checks -from fifo_in.sync_w2r.wptr\[0\]\$_DFF_PN0_/QN -to fifo_in.sync_w2r.rq1_wptr\[0\]\$_DFF_PN0_/D`

Output:

```
>>> report_checks -from fifo_in.sync_w2r.wptr\[0\]\$_DFF_PN0_/QN -to fifo_in.sync_w2r.rq1_wptr\[0\]\$_DFF_PN0_/D
[WARNING STA-0009] No common period was found between clocks clk and clk_uncore.
Startpoint: fifo_in.sync_w2r.wptr[0]$_DFF_PN0_
            (rising edge-triggered flip-flop clocked by clk_uncore)
Endpoint: fifo_in.sync_w2r.rq1_wptr[0]$_DFF_PN0_
          (rising edge-triggered flip-flop clocked by clk)
Path Group: clk
Path Type: max

  Delay    Time   Description
---------------------------------------------------------
332000.00 332000.00   clock clk_uncore (rise edge)
  40.33 332040.31   clock network delay (propagated)
   0.00 332040.31 ^ fifo_in.sync_w2r.wptr[0]$_DFF_PN0_/CLK (DFFASRHQNx1_ASAP7_75t_R)
  38.00 332078.31 v fifo_in.sync_w2r.wptr[0]$_DFF_PN0_/QN (DFFASRHQNx1_ASAP7_75t_R)
   7.33 332085.66 ^ _22035_/Y (INVx1_ASAP7_75t_R)
   0.00 332085.66 ^ fifo_in.sync_w2r.rq1_wptr[0]$_DFF_PN0_/D (DFFASRHQNx1_ASAP7_75t_R)
        332085.66   data arrival time

332000.97 332000.97   clock clk (rise edge)
 159.71 332160.69   clock network delay (propagated)
 -10.00 332150.69   clock uncertainty
   0.00 332150.69   clock reconvergence pessimism
        332150.72 ^ fifo_in.sync_w2r.rq1_wptr[0]$_DFF_PN0_/CLK (DFFASRHQNx1_ASAP7_75t_R)
 -11.71 332138.97   library setup time
        332138.97   data required time
---------------------------------------------------------
        332138.97   data required time
        -332085.66   data arrival time
---------------------------------------------------------
          53.32   slack (MET)

```

### Expected Behavior

Look at the very large delay for the clock, 332000. Where does this number come from? I would have expected something that was on the order of a clock period.

### Environment

```shell
OpenROAD v2.0-11860-g04d4677f6
```

### To Reproduce

See above

maliberty, 2024-01-26:
Try adding -format full_clock_expanded to see more details.  Have you run CTS?

oharboe, 2024-01-26:
> Try adding -format full_clock_expanded to see more details. Have you run CTS?

Yes.

Just a thought. The clocks are 333 and 1000ps, but they are not asynchronous, so 333*1000 is on the order of this surprisingly large number. Is this sognificant?

maliberty, 2024-01-27:
> Is this sognificant?

not obvious to me

oharboe, 2024-01-27:
I'm closing this for now, I forgot to include the place.odb file, so I can't re-run CTS on the example.

closed 2024-01-27 by oharboe