The-OpenROAD-Project/OpenROAD #4595: Segfault in clock tree viewer

oharboe, 2024-01-26:
### Describe the bug

1. untar https://drive.google.com/file/d/19nK2MBVsMEgIc7n-mrFOWnPK6t5RBTnx/view?usp=sharing
2. Run `./gui.sh`
3. Click around in the GUI and update clock tree viewer many times

Occasionally it crashes:

```
Signal 11 received
Stack trace:
 0# 0x0000555EF000B8A3 in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 1# 0x00007FDD59442910 in /lib/x86_64-linux-gnu/libc.so.6
 2# odb::dbNet::getITerms() in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 3# gui::DbNetDescriptor::highlight(std::any, gui::Painter&) const in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 4# gui::ClockTreeRenderer::drawTree(gui::Painter&, gui::Descriptor const*, gui::ColorGenerator&, gui::ClockTree*, int) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 5# gui::ClockTreeRenderer::drawTree(gui::Painter&, gui::Descriptor const*, gui::ColorGenerator&, gui::ClockTree*, int) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 6# gui::ClockTreeRenderer::drawTree(gui::Painter&, gui::Descriptor const*, gui::ColorGenerator&, gui::ClockTree*, int) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 7# gui::ClockTreeRenderer::drawTree(gui::Painter&, gui::Descriptor const*, gui::ColorGenerator&, gui::ClockTree*, int) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 8# gui::ClockTreeRenderer::drawTree(gui::Painter&, gui::Descriptor const*, gui::ColorGenerator&, gui::ClockTree*, int) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 9# gui::ClockTreeRenderer::drawObjects(gui::Painter&) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
10# gui::RenderThread::drawBlock(QPainter*, odb::dbBlock*, odb::Rect const&, int) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
11# gui::RenderThread::draw(QImage&, QRect const&, std::set<gui::Selected, std::less<gui::Selected>, std::allocator<gui::Selected> > const&, std::array<std::set<gui::Selected, std::less<gui::Selected>, std::allocator<gui::Selected> >, 8ul> const&, std::vector<std::unique_ptr<gui::Ruler, std::default_delete<gui::Ruler> >, std::allocator<std::unique_ptr<gui::Ruler, std::default_delete<gui::Ruler> > > > const&, double, QColor const&) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
12# gui::RenderThread::run() in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
13# 0x00007FDD5A4DAEB1 in /lib/x86_64-linux-gnu/libQt5Core.so.5
14# 0x00007FDD59497ADA in /lib/x86_64-linux-gnu/libc.so.6
15# 0x00007FDD5952847C in /lib/x86_64-linux-gnu/libc.so.6
make: *** [Makefile:1075: gui_4_cts.odb] Segmentation fault (core dumped)
```

Another

```
Signal 11 received
Stack trace:
 0# 0x00005589F6DA58A3 in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 1# 0x00007FB60D442910 in /lib/x86_64-linux-gnu/libc.so.6
 2# std::_Rb_tree_increment(std::_Rb_tree_node_base const*) in /lib/x86_64-linux-gnu/libstdc++.so.6
 3# gui::RenderThread::drawBlock(QPainter*, odb::dbBlock*, odb::Rect const&, int) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 4# gui::RenderThread::draw(QImage&, QRect const&, std::set<gui::Selected, std::less<gui::Selected>, std::allocator<gui::Selected> > const&, std::array<std::set<gui::Selected, std::less<gui::Selected>, std::allocator<gui::Selected> >, 8ul> const&, std::vector<std::unique_ptr<gui::Ruler, std::default_delete<gui::Ruler> >, std::allocator<std::unique_ptr<gui::Ruler, std::default_delete<gui::Ruler> > > > const&, double, QColor const&) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 5# gui::RenderThread::run() in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 6# 0x00007FB60E4DAEB1 in /lib/x86_64-linux-gnu/libQt5Core.so.5
 7# 0x00007FB60D497ADA in /lib/x86_64-linux-gnu/libc.so.6
 8# 0x00007FB60D52847C in /lib/x86_64-linux-gnu/libc.so.6
make: *** [Makefile:1075: gui_4_cts.odb] Segmentation fault (core dumped)
```

### Expected Behavior

No crash

### Environment

```shell
OpenROAD v2.0-11860-g04d4677f6
```

### To Reproduce

See above

maliberty, 2024-01-26:
are you modifying the design while 'clicking around'?

oharboe, 2024-01-26:
> are you modifying the design while 'clicking around'?

I didn't realize I could "modify" the design from the GUI. How do I do that? I'm just viewing things. Jumping between timing report, maybe rescaling, etc.

maliberty, 2024-01-27:
The inspector allows limited changes (eg trash can icon).  The tcl prompt allows anything

oharboe, 2024-01-27:
> The inspector allows limited changes (eg trash can icon). The tcl prompt allows anything

It crashes without that.

oharboe, 2024-01-27:
@maliberty This crashes every time(though I don't think this is the only problem).

1. Select a square, yellow below
2. Click trashcan icon, crash

```
/usr/include/c++/13/bits/stl_vector.h:1125: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = odb::dbId<odb::_dbMTerm>; _Alloc = std::allocator<odb::dbId<odb::_dbMTerm> >; reference = odb::dbId<odb::_dbMTerm>&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.
Signal 6 received
Stack trace:
 0# 0x0000563E8C4EC143 in openroad
 1# 0x00007F5DA5242910 in /lib/x86_64-linux-gnu/libc.so.6
 2# pthread_kill in /lib/x86_64-linux-gnu/libc.so.6
 3# raise in /lib/x86_64-linux-gnu/libc.so.6
 4# abort in /lib/x86_64-linux-gnu/libc.so.6
 5# 0x00007F5DA56E219F in /lib/x86_64-linux-gnu/libstdc++.so.6
 6# 0x0000563E8E93AC2B in openroad
 7# gui::DbITermDescriptor::highlight(std::any, gui::Painter&) const in openroad
 8# gui::Selected::highlight(gui::Painter&, gui::Painter::Color const&, int, gui::Painter::Color const&, gui::Painter::Brush const&) const in openroad
 9# gui::RenderThread::drawSelected(gui::Painter&, std::set<gui::Selected, std::less<gui::Selected>, std::allocator<gui::Selected> > const&) in openroad
10# gui::RenderThread::draw(QImage&, QRect const&, std::set<gui::Selected, std::less<gui::Selected>, std::allocator<gui::Selected> > const&, std::array<std::set<gui::Selected, std::less<gui::Selected>, std::allocator<gui::Selected> >, 8ul> const&, std::vector<std::unique_ptr<gui::Ruler, std::default_delete<gui::Ruler> >, std::allocator<std::unique_ptr<gui::Ruler, std::default_delete<gui::Ruler> > > > const&, double, QColor const&) in openroad
11# gui::RenderThread::run() in openroad
12# 0x00007F5DA62DAEB1 in /lib/x86_64-linux-gnu/libQt5Core.so.5
13# 0x00007F5DA5297ADA in /lib/x86_64-linux-gnu/libc.so.6
14# 0x00007F5DA532847C in /lib/x86_64-linux-gnu/libc.so.6
./gui.sh: line 7: 597387 Aborted                 (core dumped) ODB_FILE=results/asap7/mock-cpu/base/4_cts.odb openroad -gui -no_init ${SCRIPTS_DIR}/gui.tcl
```

[image]

oharboe, 2024-03-06:
@maliberty @AcKoucher Still happens:

```
OpenROAD v2.0-12456-g223f01d34 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[WARNING STA-1212] /home/oyvind/OpenROAD-flow-scripts/foo/platforms/asap7/lib/asap7sc7p5t_SIMPLE_RVT_FF_nldm_211120.lib.gz line 13178, timing group from output port.
[WARNING STA-1212] /home/oyvind/OpenROAD-flow-scripts/foo/platforms/asap7/lib/asap7sc7p5t_SIMPLE_RVT_FF_nldm_211120.lib.gz line 13211, timing group from output port.
[WARNING STA-1212] /home/oyvind/OpenROAD-flow-scripts/foo/platforms/asap7/lib/asap7sc7p5t_SIMPLE_RVT_FF_nldm_211120.lib.gz line 13244, timing group from output port.
[WARNING STA-1212] /home/oyvind/OpenROAD-flow-scripts/foo/platforms/asap7/lib/asap7sc7p5t_SIMPLE_RVT_FF_nldm_211120.lib.gz line 13277, timing group from output port.
[WARNING STA-1212] /home/oyvind/OpenROAD-flow-scripts/foo/platforms/asap7/lib/asap7sc7p5t_SIMPLE_RVT_FF_nldm_211120.lib.gz line 13310, timing group from output port.
[WARNING STA-1212] /home/oyvind/OpenROAD-flow-scripts/foo/platforms/asap7/lib/asap7sc7p5t_SIMPLE_RVT_FF_nldm_211120.lib.gz line 13343, timing group from output port.
[WARNING STA-1212] /home/oyvind/OpenROAD-flow-scripts/foo/platforms/asap7/lib/asap7sc7p5t_SIMPLE_RVT_FF_nldm_211120.lib.gz line 13376, timing group from output port.
[WARNING STA-1212] /home/oyvind/OpenROAD-flow-scripts/foo/platforms/asap7/lib/asap7sc7p5t_SIMPLE_RVT_FF_nldm_211120.lib.gz line 14772, timing group from output port.
[WARNING STA-1212] /home/oyvind/OpenROAD-flow-scripts/foo/platforms/asap7/lib/asap7sc7p5t_SIMPLE_RVT_FF_nldm_211120.lib.gz line 14805, timing group from output port.
[WARNING STA-1212] /home/oyvind/OpenROAD-flow-scripts/foo/platforms/asap7/lib/asap7sc7p5t_SIMPLE_RVT_FF_nldm_211120.lib.gz line 14838, timing group from output port.
Estimating parasitics
[WARNING GUI-0066] Heat map "IR Drop" has not been populated with data.
/usr/include/c++/13/bits/stl_vector.h:1125: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = odb::dbId<odb::_dbMTerm>; _Alloc = std::allocator<odb::dbId<odb::_dbMTerm> >; reference = odb::dbId<odb::_dbMTerm>&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.
Signal 6 received
Stack trace:
 0# 0x000055820D21B333 in openroad
 1# 0x00007F4452E42990 in /lib/x86_64-linux-gnu/libc.so.6
 2# pthread_kill in /lib/x86_64-linux-gnu/libc.so.6
 3# raise in /lib/x86_64-linux-gnu/libc.so.6
 4# abort in /lib/x86_64-linux-gnu/libc.so.6
 5# 0x00007F44532E219F in /lib/x86_64-linux-gnu/libstdc++.so.6
 6# 0x000055820F67633B in openroad
 7# gui::DbITermDescriptor::highlight(std::any, gui::Painter&) const in openroad
 8# gui::Selected::highlight(gui::Painter&, gui::Painter::Color const&, int, gui::Painter::Color const&, gui::Painter::Brush const&) const in openroad
 9# gui::RenderThread::drawSelected(gui::Painter&, std::set<gui::Selected, std::less<gui::Selected>, std::allocator<gui::Selected> > const&) in openroad
10# gui::RenderThread::draw(QImage&, QRect const&, std::set<gui::Selected, std::less<gui::Selected>, std::allocator<gui::Selected> > const&, std::array<std::set<gui::Selected, std::less<gui::Selected>, std::allocator<gui::Selected> >, 8ul> const&, std::vector<std::unique_ptr<gui::Ruler, std::default_delete<gui::Ruler> >, std::allocator<std::unique_ptr<gui::Ruler, std::default_delete<gui::Ruler> > > > const&, double, QColor const&) in openroad
11# gui::RenderThread::run() in openroad
12# 0x00007F4453EDAEB1 in /lib/x86_64-linux-gnu/libQt5Core.so.5
13# 0x00007F4452E97B5A in /lib/x86_64-linux-gnu/libc.so.6
14# 0x00007F4452F285FC in /lib/x86_64-linux-gnu/libc.so.6
./gui.sh: line 7: 125556 Aborted                 (core dumped) ODB_FILE=results/asap7/mock-cpu/base/4_cts.odb openroad -gui -no_init ${SCRIPTS_DIR}/gui.tcl
```

oharboe, 2024-03-06:
@AcKoucher @maliberty Repeatable procedure:

1. Select a rectangle
2. Update clock tree
3. Go to "Inspector", click trash can icon.

[image]

```
[WARNING GUI-0066] Heat map "IR Drop" has not been populated with data.
/usr/include/c++/13/bits/stl_vector.h:1125: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = odb::dbId<odb::_dbMTerm>; _Alloc = std::allocator<odb::dbId<odb::_dbMTerm> >; reference = odb::dbId<odb::_dbMTerm>&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.
Signal 6 received
Stack trace:
 0# 0x0000560E7F754333 in openroad
 1# 0x00007F8D02242990 in /lib/x86_64-linux-gnu/libc.so.6
 2# pthread_kill in /lib/x86_64-linux-gnu/libc.so.6
 3# raise in /lib/x86_64-linux-gnu/libc.so.6
 4# abort in /lib/x86_64-linux-gnu/libc.so.6
 5# 0x00007F8D026E219F in /lib/x86_64-linux-gnu/libstdc++.so.6
 6# 0x0000560E81BAF33B in openroad
 7# odb::dbITerm::getName[abi:cxx11](char) const in openroad
 8# gui::DbITermDescriptor::getName[abi:cxx11](std::any) const in openroad
 9# gui::SelectionModel::data(QModelIndex const&, int) const in openroad
10# QSortFilterProxyModel::data(QModelIndex const&, int) const in /lib/x86_64-linux-gnu/libQt5Core.so.5
11# QStyledItemDelegate::initStyleOption(QStyleOptionViewItem*, QModelIndex const&) const in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
12# QStyledItemDelegate::sizeHint(QStyleOptionViewItem const&, QModelIndex const&) const in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
13# 0x00007F8D0443E316 in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
14# QTableView::sizeHintForRow(int) const in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
15# 0x00007F8D043F8F13 in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
16# QHeaderView::visualIndexAt(int) const in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
17# QTableView::sizeHintForColumn(int) const in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
18# 0x00007F8D043F8F13 in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
19# QHeaderView::length() const in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
20# QTableView::updateGeometries() in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
21# QAbstractItemView::doItemsLayout() in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
22# QTableView::doItemsLayout() in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
23# QAbstractItemView::timerEvent(QTimerEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
24# QObject::event(QEvent*) in /lib/x86_64-linux-gnu/libQt5Core.so.5
25# QFrame::event(QEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
26# QApplicationPrivate::notify_helper(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
27# QCoreApplication::notifyInternal2(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Core.so.5
28# QTimerInfoList::activateTimers() in /lib/x86_64-linux-gnu/libQt5Core.so.5
29# 0x00007F8D03527A2C in /lib/x86_64-linux-gnu/libQt5Core.so.5
30# 0x00007F8D02513B2C in /lib/x86_64-linux-gnu/libglib-2.0.so.0
31# 0x00007F8D0256F46F in /lib/x86_64-linux-gnu/libglib-2.0.so.0
32# g_main_context_iteration in /lib/x86_64-linux-gnu/libglib-2.0.so.0
33# QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) in /lib/x86_64-linux-gnu/libQt5Core.so.5
34# QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) in /lib/x86_64-linux-gnu/libQt5Core.so.5
35# QCoreApplication::exec() in /lib/x86_64-linux-gnu/libQt5Core.so.5
36# gui::startGui(int&, char**, Tcl_Interp*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool) in openroad
37# ord::tclAppInit(Tcl_Interp*) in openroad
38# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
39# main in openroad
40# 0x00007F8D02228150 in /lib/x86_64-linux-gnu/libc.so.6
41# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
42# _start in openroad
./gui.sh: line 7: 125869 Aborted                 (core dumped) ODB_FILE=results/asap7/mock-cpu/base/4_cts.odb openroad -gui -no_init ${SCRIPTS_DIR}/gui.tcl
```

maliberty, 2024-04-22:
The image link is broken.  I tried your steps above but got no crash.

maliberty, 2024-04-23:
The link worked on reload but the problem still doesn't reproduce

oharboe, 2024-04-23:
Still crashes:

```
$ ./gui.sh 
QSocketNotifier: Can only be used with threads started with QThread
OpenROAD v2.0-13363-g5aef95740 
Features included (+) or not (-):  +Charts +GPU +GUI +Python
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
Estimating parasitics
/usr/include/c++/13/bits/stl_vector.h:1125: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = odb::dbId<odb::_dbMTerm>; _Alloc = std::allocator<odb::dbId<odb::_dbMTerm> >; reference = odb::dbId<odb::_dbMTerm>&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.
Signal 6 received
Stack trace:
 0# 0x00006095877B1533 in openroad
 1# 0x000072E673C42990 in /lib/x86_64-linux-gnu/libc.so.6
 2# pthread_kill in /lib/x86_64-linux-gnu/libc.so.6
 3# raise in /lib/x86_64-linux-gnu/libc.so.6
 4# abort in /lib/x86_64-linux-gnu/libc.so.6
 5# 0x000072E6740E219F in /lib/x86_64-linux-gnu/libstdc++.so.6
 6# 0x0000609589CE5B4B in openroad
 7# gui::DbITermDescriptor::highlight(std::any, gui::Painter&) const in openroad
 8# gui::Selected::highlight(gui::Painter&, gui::Painter::Color const&, int, gui::Painter::Color const&, gui::Painter::Brush const&) const in openroad
 9# gui::RenderThread::drawSelected(gui::Painter&, std::set<gui::Selected, std::less<gui::Selected>, std::allocator<gui::Selected> > const&) in openroad
10# gui::RenderThread::draw(QImage&, QRect const&, std::set<gui::Selected, std::less<gui::Selected>, std::allocator<gui::Selected> > const&, std::array<std::set<gui::Selected, std::less<gui::Selected>, std::allocator<gui::Selected> >, 8ul> const&, std::vector<std::unique_ptr<gui::Ruler, std::default_delete<gui::Ruler> >, std::allocator<std::unique_ptr<gui::Ruler, std::default_delete<gui::Ruler> > > > const&, double, QColor const&) in openroad
11# gui::RenderThread::run() in openroad
12# 0x000072E674CDAEB1 in /lib/x86_64-linux-gnu/libQt5Core.so.5
13# 0x000072E673C97B5A in /lib/x86_64-linux-gnu/libc.so.6
14# 0x000072E673D285FC in /lib/x86_64-linux-gnu/libc.so.6
./gui.sh: line 7: 2081078 Aborted                 (core dumped) ODB_FILE=results/asap7/mock-cpu/base/4_cts.odb openroad -gui -no_init ${SCRIPTS_DIR}/gui.tcl
```

oharboe, 2024-04-23:
[image]

oharboe, 2024-04-23:
@maliberty So here is my best guess:

- a render is triggered and the selected/highlighted list is duplicated for the low priority rendering thread
- the main thread, higher priority, preempts rendering and deletes the elements on the selected/highlight list
- the low priority rendering thread resumes and is now accessing deleted elements

I looked for, but could not find in RenderThread.cc:

- a method to abort the current rendering and wait for it to complete
- this method should be called before deleting/modifying what is being rendered