The-OpenROAD-Project/OpenROAD #4612: Detail route has long pause after 100% in nth iterations

oharboe, 2024-02-01:
_Originally posted by @oharboe in https://github.com/The-OpenROAD-Project/OpenROAD/discussions/4598_

To reproduce

1. untar 
2. Run `NUM_CORES=16 ./run-me-TilePRCIDomain-asap7-base.sh`
3. Observe long pause after 100% below

```
[ca. 4 hours or so]
    Completing 100% with 79991 violations.
    elapsed time = 04:47:55, memory = 59083.30 (MB).
[several hours]
[INFO DRT-0199]   Number of violations = 87967.
[continued output]
```

maliberty, 2024-02-01:
I know the connectivity checker runs at that point but I'm not sure if that's the issue

oharboe, 2024-02-01:
> I know the connectivity checker runs at that point but I'm not sure if that's the issue

Would it make sense to add some idempotent progress output that is displayed if this takes much longer than normal?

That would allow deltaDebug to be used....

maliberty, 2024-02-01:
How would you define 'much longer than normal' in an idempotent way?

oharboe, 2024-02-01:
> How would you define 'much longer than normal' in an idempotent way?

Good question. I don't even know what the code is doing at this point and it requires an understanding of what the code is doing and how to quantify progress in an idempotent way that is independent of number of cores running...

maliberty, 2024-02-01:
As you asking for a progress message or a "more than normal" message?

oharboe, 2024-02-01:
> As you asking for a progress message or a "more than normal" message?

For deltaDebug a "more than normal" would do as long as it reduces the test case to something practical... Still a tricky requirement.

maliberty, 2024-02-01:
I don't see any way to measure time in an idempotent way - it will vary too much.  I also don't know what normal is.  We could add a progress message.

oharboe, 2024-02-01:
> I don't see any way to measure time in an idempotent way - it will vary too much. I also don't know what normal is. We could add a progress message.

Maybe initially add some debug logging?

How else to debug something that takes ca 4 hours to reach?

maliberty, 2024-02-01:
Sure though that won't be idempotent

oharboe, 2024-02-01:
> Sure though that won't be idempotent

Less critical that it is not noisy, so easier to make ot idempotent?

QuantamHD, 2024-02-01:
+1 to preserving hermiticity/idempotency as a top priority. We should just have an option to turn on progress messages/debug output.

oharboe, 2024-02-01:
> +1 to preserving hermiticity/idempotency as a top priority. We should just have an option to turn on progress messages/debug output.

Indeed. 

Other than that, the concern is that something has to drive deltaDebug robustly.

osamahammad21, 2024-08-31:
It seems that there is one extremely large net that needs to be partitioned in my opinion(235K pins). The net is causing throttling during operations that are performed on each net individually, such as connectivity checking and guide processing.

[image]

oharboe, 2024-08-31:
I think this is a clock that was omitted in .sdc(out of ignorance by me).

The other massive fanout net is for synchronous reset.

osamahammad21, 2024-08-31:
Still it seems to me that there should be a warning in DRT to detect such case and warn the user instead of leaving them in the dark for hours. DRT should eventually be able to route any net no matter how large it is. So there is nothing technically wrong in having this large net for DRT but it definitely makes its job much harder.

oharboe, 2024-08-31:
I think a report of the highest fanout nets immediately after synthesis would cover this, no?

maliberty, 2024-08-31:
No as we may buffer such a net later in the flow with repair_design/timing.

oharboe, 2024-08-31:
The point is just to catch nets that should have been clocks, large fanout resets, etc.

osamahammad21, 2024-08-31:
@oharboe There are warning all over the logs with this net being skipped previously by the resizer for having high fanout.
```
[WARNING RSZ-0069] skipping net auto_tap_clock_in_clock with 235689 pins.
[WARNING RSZ-0069] skipping net net4 with 235689 pins.
```
As matt mentioned, it should have been handled, or even complained about, before reaching the detailed router. But there is a limit on the net fanout handled by rsz. If it is exceeded, the net is ignore by rsz. We'll see if this limit could be removed.

closed 2024-08-31 by maliberty

maliberty, 2024-08-31:
> The point is just to catch nets that should have been clocks, large fanout resets, etc.

No nets should be high fanout when reaching drt.

oharboe, 2024-08-31:
Can/should a forgotten clock be caught earlier?

Should this not be seen easily at check_setup time in floorplan.tcl?

A list of 10 highest fanout nets at check_setup could be useful?

maliberty, 2024-08-31:
Its not a violation to have a high fanout net early in the flow so I don't see it belonging in check_setup.  https://github.com/The-OpenROAD-Project/OpenROAD/pull/5676 is a reasonable solution if the runtime is acceptable.

oharboe, 2024-08-31:
> Its not a violation to have a high fanout net early in the flow so I don't see it belonging in check_setup. #5676 is a reasonable solution if the runtime is acceptable.

I meant: isn't this a problem that can be seen that early in the flow?

What reports exist today that the user can look at to discover this as early as after check_setup?

The use-case I'm interested in is where I didn't intend to have such a net, it should have been a clock. So the flow "fixing" it later is just making my mistake more obscured.

maliberty, 2024-09-01:
There is no easy way to know what you intended.  You can set a max fanout and `report_check_types  -max_fanout` will tell you of the violations.