The-OpenROAD-Project/OpenROAD #4618: openroad needs 25GB to load ispd24 ariane133_51 with 120k std cells

stefanottili, 2024-02-03:
### Describe the bug

openroad needs 25GB to just read/display the smallest of the ispd24 test cases, ariane133_51.def with 120k std cells.

The placement density map looks nice, but 'I think I need a bigger boat' to be able to have a look at the other test cases.

I just recently ran the ispd19_test6 detailed routing with 178K std cells without issues on my 16GB M1, so why does OR all the sudden need 25GB to display 120k placed cells ?

cluster-001.def with 9.8M std cells uses 10GB after read def, but then it keeps allocating memory

### Expected Behavior

a) use less then 25GB to display 120k placed std cells with placement density map
b) give some feedback on how much memory the tool is allocating

### Environment

```shell
OpenROAD v2.0-12068-ge92b2439a as of Feb 3rd 2024

MacOS M1 16GB
```

### To Reproduce

openroad -gui ispd24.or
```
read_lef ../chips/ispd24/lef/Nangate.lef
read_def ../chips/ispd24/def/ariane133_68.def.gz
#read_def ../chips/ispd24/def/ariane133_51.def.gz
#read_def ../chips/ispd24/def/mempool_tile.def.gz
#read_def ../chips/ispd24/def/nvdla.def.gz
#read_def ../chips/ispd24/def/bsg_chip.def.gz
#read_def ../chips/ispd24/def/mempool_group.def.gz
#read_def ../chips/ispd24/def/cluster-001.def.gz
```
https://liangrj2014.github.io/ISPD24_contest/

### Relevant log output

```shell
[INFO ODB-0222] Reading LEF file: ../chips/ispd24/lef/Nangate.lef
[INFO ODB-0223]     Created 22 technology layers
[INFO ODB-0224]     Created 27 technology vias
[INFO ODB-0225]     Created 148 library cells
[INFO ODB-0226] Finished LEF file:  ../chips/ispd24/lef/Nangate.lef
[INFO ODB-0127] Reading DEF file: ../chips/ispd24/def/ariane133_68.def.gz
[INFO ODB-0128] Design: ariane
[INFO ODB-0094] 		Created 100000 Insts
[INFO ODB-0097] 		Created 100000 Nets
[INFO ODB-0130]     Created 435 pins.
[INFO ODB-0131]     Created 120202 components and 732255 component-terminals.
[INFO ODB-0133]     Created 127026 nets and 432355 connections.
[INFO ODB-0134] Finished DEF file: ../chips/ispd24/def/ariane133_68.def.gz
```

### Screenshots

[image]

stefanottili, 2024-02-03:
Just to put things into perspective:
As recent as 1997 one could do P&R of 150k std cells using 4 layers of metal on a $12k 100MHz SparcStation 20.
I remember it vividly, since the P&R tool at the time started swapping during routing with "only" 128MB.
We had to buy an additional 128MB of memory for $$$$.

stefanottili, 2024-02-04:
Further testing reveals that displaying the placement density requires only 500MB, much more in line with expectations.
Switching on the "Estimated Congestion Heat Map" then makes the memory requirement balloon to 25.5GB

maliberty, 2024-02-04:
The rudy code is quite new and could well have an issue.

maliberty, 2024-02-04:
@gudeh please take a look

stefanottili, 2024-02-04:
On a previous OR run, I had a look at the Rudy map.
Since the heat map display settings are persistent, after starting a new run and loading lef/def, Rudy was run without me being aware of that.
It would be helpful if running Rudy would print something to the console and also report runtime and memory usage.

QuantamHD, 2024-02-04:
Also this visualizer is supposed to be fast. I don't think adding additional logging is strictly necessary just because in this case it's misbehaving

gudeh, 2024-02-05:
@stefanottili Hi, thanks for the detailed description of the problem. Could you please explain how are you measuring the memory usage exactly?

gudeh, 2024-02-05:
I tested with a few designs (ng45/gcd, ng45/tinyRocket, and 130hd/microwatt), with OR version of both before and after my recent changes to RUDY  and I noticed no sudden spikes on memory usage. I will try to load the design you pointed out.

stefanottili, 2024-02-05:
I’m using a 16GB MacOS AirBook M1 (arm64) and its resource monitor, which shows that it starts swapping displaying RUDY on ariane133_51. The other ispd24 testcases run out of memory and OR gets killed.Only displaying the placement map works for all testcases.

gadfort, 2024-02-05:
Looking at the GCells: X is: 0 190 3990 7790 11590 15390 19190 22990 26790 30590 34390 38190 41990 (RUDY is assuming the first 2 are the "normal" size the GRT heatmap probably does the same). If the GCellGrid was 0 3800 ... it would be fine.

gudeh, 2024-02-05:
@stefanottili I could not really load the exact design from LEF and DEF files in the link you provided. I assume it is due to the "ispd24.or" in your command, is this supposed to be run with docker?

maliberty, 2024-02-05:
Is the gcell grid coming from another global router outside OR?

gadfort, 2024-02-05:
@gudeh  it was a little bit of a scavenger hunt but I found the files here:
https://drive.google.com/drive/folders/1ocChoQupNxlLBH2hgqkPwR0-7D3ocTwm
@maliberty outside of OR, directly from the DEF call

gadfort, 2024-02-05:
```
[[ord::get_db_block] getGCellGrid] getGridX
[[ord::get_db_block] getGCellGrid] getGridY
```

maliberty, 2024-02-05:
I recall drt has some code to figure out the gcell size by finding the most common value.  I think that was to deal with contest guides as well.

gadfort, 2024-02-05:
@maliberty this works for me: https://github.com/The-OpenROAD-Project/OpenROAD/pull/4621
But does not address GRT (although it looks like it might create a new set?)

gadfort, 2024-02-05:
GRTs code: https://github.com/The-OpenROAD-Project/OpenROAD/blob/80fd75d38f5f0c6a71d4c50d69bbcf2cb956db15/src/grt/src/heatMap.cpp#L158

gudeh, 2024-02-05:
It seems GRT is also consuming too much memory.

closed 2024-02-06 by maliberty

stefanottili, 2024-02-06:
RUDY run's very fast and I was able to run global route for the ispd24/def/mempool_group.def with 3M std cell using 40GB or memory.

PR #4621 (2024-02-06): gui: allow for non-uniform grids in heatmaps and apply gcells to heatmaps
Changes:
- allow for heatmaps to have non-uniform grids
- add reset to RUDY calculation, otherwise each call just sums on the previous
- add db callbacks to RUDY heatmap to update when instances move
- add mutex to heatmap `ensureMap` to prevent RenderThread from calling it more than once at a time

Closes: #4618