The-OpenROAD-Project/OpenROAD #4623: Improve quality of results of MegaBoom design

oharboe, 2024-02-06:
### Description

After learning more about Chipyard+MegaBoom and how it is built with [commercial tools](https://carrv.github.io/2020/papers/CARRV2020_paper_15_Zhao.pdf).

Here is the order to things to improve quality of results best as I know:

1. Build BoomTile with [set_clock_latency](https://github.com/The-OpenROAD-Project/megaboom/blob/95711fcc65ae9fd92f76d53b797e5e0e896fd9f1/constraints-boomtile.sdc#L15) roughly equal to average clock insertion latency. Single clock.
2. Set synchronous reset to [false path](https://github.com/The-OpenROAD-Project/megaboom/blob/95711fcc65ae9fd92f76d53b797e5e0e896fd9f1/constraints-boomtile.sdc#L24), for now.
3. Flatten BoomTile and get macro placement to do a decent job without hierarchical analysis, will certainly require macro placement constraints file. Syntax TBD.
4. Build L2 cache as a macro in the same way as BoomTile, with source latency.
5. Update clock tree synthesis to take macro insertion latency into account
6. Build ChipTop flattened. There is an asynchronous FIFO crossing between BoomTile and L2, which both connects to a system bus.  With this asynchronous FIFO, the skew of the clock tree insertion latency of BoomTile and L2 won't matter w.r.t. the ChipTop pins.

Some [screenshots](https://github.com/The-OpenROAD-Project/megaboom/commit/95711fcc65ae9fd92f76d53b797e5e0e896fd9f1) from `bazelisk build BoomTile_route BoomTile_route_orfs` and `bazel-bin/logs/asap7/BoomTile/base/route.sh gui_route`:

[image]

oharboe, 2024-02-06:
@rovinski @tspyrou @maliberty @louiic @precisionmoon I thought it was useful to write up a feature request that gives some context to the individual improvements in the pipeline.

maliberty, 2024-02-06:
Other than the items that are already open I'm not clear what you are requesting.  Retiming is a large task and would be a yosys task so you would have to get their interest in it.

oharboe, 2024-02-06:
> Other than the items that are already open I'm not clear what you are requesting. Retiming is a large task and would be a yosys task so you would have to get their interest in it.

There is no request here, but I thought it would be useful to have a public writeup that ties together why these individual improvements are requested.

closed 2024-03-06 by oharboe