The-OpenROAD-Project/OpenROAD #4629: openroad uses 10GB reading ispd24/cluster-001.def, openroad -gui requires 44GB

stefanottili, 2024-02-06:
### Describe the bug

"openroad ispd24.or" allocates 10GB to read ispd24 cluster-001.def with 9.8M placed std cells.
"openroad -gui ispd24.or" requires 44GB ... 34GB for rTree's ?

### Expected Behavior

openroad -gui shouldn't need that much memory to display placement

### Environment

```shell
OpenROAD v2.0-12114-g0bd67a2b0 (Feb 6 2024) M1 macOS Sonoma
```

### To Reproduce

openroad -gui ispd24.or
```
read_lef lef/Nangate.lef
read_def def/cluster-001.def.gz
```
https://drive.google.com/drive/folders/1ocChoQupNxlLBH2hgqkPwR0-7D3ocTwm

maliberty, 2024-02-06:
There is no def/cluster-001.def.gz in the gdrive link

maliberty, 2024-02-06:
I suppose you meant def/cluster.def

stefanottili, 2024-02-06:
yes.

For the other test cases, even the 3M std cell mempool_group, the memory increase due to -gui is not as pronounced.

maliberty, 2024-02-06:
Its not the rtrees it the hierarchy browser.  @gadfort thoughts?  Lazy load? Don't include the leaf non-macro instances?

gadfort, 2024-02-06:
@maliberty  there are data classes in Qt that would support that nicely. I looked at it way back when, but wasn't able to get anything working. 

As a temporary solution, they can turn off the browser and close openroad that will stop it from opening with it again.

gadfort, 2024-02-06:
It's also possible there is a buggy implementation...

stefanottili, 2024-02-06:
Displaying design hierarchy using Qt tree views work well with klayout.

maliberty, 2024-02-06:
Does it show a tree of 9M objects?  We could optimize the interface to the db to avoid copying but I wonder how useful it is to have a huge list of all those instances.  Does it serve a helpful purpose of would eliding them be simpler?

maliberty, 2024-02-06:
[image]

maliberty, 2024-02-06:
or having a few summary lines like

Buffers       XXX
Sequential YYY
Combinational ZZZ

The categories could be similar to the instance display controls

stefanottili, 2024-02-06:
lazy loading is the way to go, but scrolling thru 9.8M stdcells is certainly pointless.
The netlist info would have to be reduced to something useful.

A) use the hierarchy delimiter into to make this a hierarchy browser

B) display it grouped by sequential/combinatorial cells and “merge” sequential names, e.g. 64 ff[0], ff[1], … ff[63] into a single entry ff[0:63]

maliberty, 2024-02-07:
We don't try to reverse engineer the hierarchy from DEF names - we assume a hierarchical Verilog input if you want hierarchy.  I think a summary makes more sense and then lazy loading is a non-issue.

Another workaround
```
diff --git a/src/gui/src/browserWidget.cpp b/src/gui/src/browserWidget.cpp
index 807fb7fef..c179bbb56 100644
--- a/src/gui/src/browserWidget.cpp
+++ b/src/gui/src/browserWidget.cpp
@@ -438,7 +438,7 @@ BrowserWidget::ModuleStats BrowserWidget::populateModule(odb::dbModule* module,
   for (auto* inst : module->getInsts()) {
     insts.push_back(inst);
   }
-  stats += addInstanceItems(insts, "Leaf instances", parent, false);
+  stats += addInstanceItems(insts, "Leaf instances", parent, true);
 
   return stats;
 }
```

stefanottili, 2024-02-07:
This workaround works in that OR doesn't allocate excessive amounts of memory.
Now one can only see the first couple of hundred std cells in the hierarchy browser.

@gadfort
> As a temporary solution, they can turn off the browser and close openroad that will stop it from opening with it again.
nope, this doesn't work, I didn't have the hierarchy browser visible, but OR still used all the memory.

maliberty, 2024-02-08:
Macros can be expanded to show individual elements as there aren't that many but all others just show a summary.

PR #4632 (2024-02-08): gui: only add rows for BLOCKs to the hierarchy browser to save memory
Having millions of rows for standard cells takes too much memory and isn't very usable.

Fixes #4629
commit ccf3172e (2024-02-08): gui: only add rows for BLOCKs to the hierarchy browser to save memory

Having millions of rows for standard cells takes too much memory
and isn't very usable.

Fixes #4629
closed 2024-02-08 by maliberty