The-OpenROAD-Project/OpenROAD #4630: display "busy" cursor while RUDY is calculating the map

stefanottili, 2024-02-07:
### Describe the bug

For ispd24 cluster.def (9.8M std cell) it takes RUDY ~30 seconds to compute the map.

### Expected Behavior

Please change the cursor from "pointer" to "busy" while computing RUDY to give the user feedback that OR is doing something.

### Environment

```shell
Feb 6 OR M1 macOS Sonoma
```

### To Reproduce

openroad -gui 
read_lef lef/Nangate.lef
read_def def/cluster.def.gz

show RUDY map

maliberty, 2024-02-07:
@gudeh

gudeh, 2024-02-07:
References: 
https://liangrj2014.github.io/ISPD24_contest/
#4618

gudeh, 2024-02-12:
I noticed that other heatmap calculations also do not indicate when they are busy, and these heatmaps take some time with larger designs as well. Therefore, I have made modifications to show a busy indicator for any heatmap calculation, not just for RUDY.

gudeh, 2024-02-13:
Covered by #4638.

closed 2024-02-13 by gudeh