The-OpenROAD-Project/OpenROAD #4633: Add feature to click on bucket in slack endpoint histogram to navigate to timing report

oharboe, 2024-02-07:
### Description

It would be nice to be able to click in the endpoint slack histogram and read in those endpoints in the timing report:

[image]

### Suggested Solution

Click in endpoint histogram, read in those end points in the timing report.

oharboe, 2024-04-11:
> My concern here is what happens if you click on a very large bin (eg 7k endpoints above). That could get rather slow to generate in the second model.

You can still respect the filtering option which defaults to maximum 50 endpoints.

The use-case of right clicking in the histogram on something interesting is to see some of the worst endpoints in that range.

maliberty, 2024-04-11:
That seems reasonable.  @AcKoucher does that answer the question?

PR #4955 (2024-04-22): gui: timing report based on slack histogram
Resolve #4633
closed 2024-04-22 by eder-matheus