The-OpenROAD-Project/OpenROAD #4634: CTS segfault

oharboe, 2024-02-07:
A segfault.

To reproduce:

1. untar [cts_L1MetadataArray_asap7_test_2024-02-07_08-32.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/14190467/cts_L1MetadataArray_asap7_test_2024-02-07_08-32.tar.gz)
2. `./run-me-L1MetadataArray-asap7-test.sh`

Output:

```
OpenROAD v2.0-12049-gb0d81bbb0 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
clock_tree_synthesis -sink_clustering_enable -balance_levels -distance_between_buffers 60
[INFO CTS-0050] Root buffer is BUFx24_ASAP7_75t_R.
[INFO CTS-0051] Sink buffer is BUFx24_ASAP7_75t_R.
[INFO CTS-0052] The following clock buffers will be used for CTS:
                    BUFx24_ASAP7_75t_R
[INFO CTS-0049] Characterization buffer is BUFx24_ASAP7_75t_R.
[INFO CTS-0007] Net "clock" found for clock "clock".
Signal 11 received
Stack trace:
 0# 0x0000562ECFC293F3 in openroad
 1# 0x00007F0A69242990 in /lib/x86_64-linux-gnu/libc.so.6
 2# sta::ConcreteCell::findPort(char const*) const in openroad
 3# cts::TritonCTS::separateMacroRegSinks(odb::dbNet*&, cts::Clock&, std::unordered_set<odb::dbMaster*, std::hash<odb::dbMaster*>, std::equal_to<odb::dbMaster*>, std::allocator<odb::dbMaster*> > const&, std::vector<std::pair<odb::dbInst*, odb::dbMTerm*>, std::allocator<std::pair<odb::dbInst*, odb::dbMTerm*> > >&, std::vector<std::pair<odb::dbInst*, odb::dbMTerm*>, std::allocator<std::pair<odb::dbInst*, odb::dbMTerm*> > >&) in openroad
 4# cts::TritonCTS::initClockTreeForMacrosAndRegs(odb::dbNet*&, std::unordered_set<odb::dbMaster*, std::hash<odb::dbMaster*>, std::equal_to<odb::dbMaster*>, std::allocator<odb::dbMaster*> > const&, cts::Clock&, cts::TreeBuilder*) in openroad
 5# cts::TritonCTS::initClock(odb::dbNet*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, cts::TreeBuilder*) in openroad
 6# cts::TritonCTS::initOneClockTree(odb::dbNet*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, cts::TreeBuilder*) in openroad
 7# cts::TritonCTS::populateTritonCTS() in openroad
 8# cts::TritonCTS::runTritonCts() in openroad
 9# 0x0000562ED02D1F39 in openroad
10# TclNRRunCallbacks in /lib/x86_64-linux-gnu/libtcl8.6.so
11# 0x00007F0A6DB08C43 in /lib/x86_64-linux-gnu/libtcl8.6.so
12# Tcl_EvalEx in /lib/x86_64-linux-gnu/libtcl8.6.so
13# Tcl_Eval in /lib/x86_64-linux-gnu/libtcl8.6.so
14# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in openroad
15# ord::tclAppInit(Tcl_Interp*) in openroad
16# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
17# main in openroad
18# 0x00007F0A69228150 in /lib/x86_64-linux-gnu/libc.so.6
19# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
20# _start in openroad
./run-me-L1MetadataArray-asap7-test.sh: line 7: 910602 Segmentation fault      (core dumped) openroad -no_init ${SCRIPTS_DIR}/cts.tcl
```

oharboe, 2024-02-07:
@precisionmoon FYI, #4607 is not needed to reproduce this problem.

precisionmoon, 2024-02-07:
@oharboe, thanks much for bringing this to my attention.   You are correct.  The issue is not related to the new insertion delay enhancement.   Some APIs need to be bullet proofed.  The fix has been checked into https://github.com/The-OpenROAD-Project/OpenROAD/pull/4607.

precisionmoon, 2024-02-09:
I'm closing this since PR #4607 has been merged.

closed 2024-02-09 by precisionmoon