The-OpenROAD-Project/OpenROAD #4644: report_clock_skew does not take macro insertion latency into account

oharboe, 2024-02-09:
### Describe the bug

1. untar [gui.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/14218067/gui.tar.gz)
2. `./run-me-mock-array-asap7-base.sh`

[image]

### Expected Behavior

The skew should take macro insertion latency into account

### Environment

```shell
OpenROAD v2.0-12142-g1b0a9b9c7
```

### To Reproduce

See above

maliberty, 2024-02-09:
@tspyrou for Cherry

tspyrou, 2024-02-09:
@oharboe virtual clocks do not have a clock tree and won't have a reported skew.

tspyrou, 2024-02-09:
We set all clocks to be propagated here. The virtual ones will receive the warning. We could leave this to users to set but we have had cases of pilot error where users forget.
https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/617eb260fd18b5b8260e3089580e3b4549ffdaba/flow/scripts/gui.tcl#L44

oharboe, 2024-02-09:
@tspyrou I can, manually, figure/reason out the skew of the WNS path.

More sophisticated skew metrics would figure out how much performance is left on the table. As Austin said: zero skew is not ideal skew.

tspyrou, 2024-04-02:
This was implemented in a commit on Feb 27th. Sorry for not updating until now.

closed 2024-04-02 by tspyrou

oharboe, 2024-04-02:
@tspyrou Thanks! I tried out the feature and I can see it implemented, but I'm a bit puzzled by what I see: https://github.com/The-OpenROAD-Project/OpenROAD/issues/4886