The-OpenROAD-Project/OpenROAD #4646: Timing Report should report the top N WNS paths

oharboe, 2024-02-09:
### Describe the bug

1. untar  [timing.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/14220733/timing.tar.gz)
2. run `./run-me-mock-array-asap7-base.sh`

[image]

[image]

### Expected Behavior

I would expect that the top of the list should be identical as it is sorted on ascending WNS

### Environment

```shell
OpenROAD v2.0-12143-g4aabf0d3c
```

### To Reproduce

See above

maliberty, 2024-03-17:
In text mode you can get a sense of the issue

```
openroad> report_checks -format summary -endpoint_count 5 -path_group clock
Startpoint                           Endpoint                               Slack
--------------------------------------------------------------------------------
ces_6_1/io_lsbOuts_7 (Element)       ces_6_5/io_lsbIns_4 (Element)         -10.96
ces_6_0/io_lsbOuts_7 (Element)       ces_6_4/io_lsbIns_4 (Element)         -10.72
ces_7_0/io_lsbOuts_7 (Element)       ces_7_4/io_lsbIns_4 (Element)         -10.64
ces_7_1/io_lsbOuts_7 (Element)       ces_7_5/io_lsbIns_4 (Element)         -10.62
ces_6_1/io_lsbOuts_7 (Element)       ces_6_5/io_lsbIns_4 (Element)         -10.10

openroad> report_checks -format summary -endpoint_count 5 -path_group clock_vir
Startpoint                           Endpoint                               Slack
--------------------------------------------------------------------------------
ces_7_0/io_outs_left[0] (MockArray)  io_outs_left_7[0] (output)            149.64
ces_7_0/io_outs_up[22] (MockArray)   io_outs_up_0[22] (output)             149.92
ces_7_0/io_outs_left[0] (MockArray)  io_outs_left_7[0] (output)            150.61
ces_7_0/io_outs_up[38] (MockArray)   io_outs_up_0[38] (output)             151.21
ces_7_0/io_outs_up[36] (MockArray)   io_outs_up_0[36] (output)             151.55
```

There are two different clock groups here.  I believe it is trying to show at least one path from each group.  There is currently no option equivalent to -path_group in the GUI.  I think the current behavior is ok but we might add such an option.  Would that resolve your concern?

oharboe, 2024-03-18:
@maliberty Thanks for explaining. I don't think there is anything to fix here then, as such. The problem is that I "impostering until I make it". It would be nice if the GUI was explaining things in a way such that EDA novices such as me don't get confused, but adding options I don't know what means and I don't know how to use isn't going to lower the barrier...

closed 2024-03-18 by oharboe

maliberty, 2024-03-18:
It wasn't obvious to me either fwiw.