The-OpenROAD-Project/OpenROAD #4647: Pins swapped incorrectly on and2b cell

MichaelBell, 2024-02-09:
### Describe the bug

During the Resizer Timing step of the flow for my Tiny Tapeout design, the inputs of a `sky130_fd_sc_hd__and2b_1` cell were swapped.

The function of the cell is !A & B, so swapping the inputs changes behaviour of the design.

### Expected Behavior

The resizer should not swap pins for cells where the behaviour of the design would be changed by the swap.

### OpenROAD Environment

```shell
Using the docker image: efabless/openlane-tools:openroad_app-75f2f325b7a42e56a92404f33af8e96530d9b202-centos-7-amd64
```

### OpenLane Environment

```shell
open_pdks cd1748bb197f9b7af62a54507de6624e30363943
Kernel: Linux v5.15.133.1-microsoft-standard-WSL2
Distribution: ubuntu 22.04
Python: v3.10.12 (OK)
Container Engine: docker v20.10.25 (OK)
OpenLane Git Version: 9dbd8b5ea2bd891bed4dcc97df5c7439083f0368
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

9dbd8b5 2024-01-11T15:26:56+02:00 Fix `-synth_explore` crash (#2085) - Kareem Farid -  (grafted, HEAD)
---
Git Remotes

origin  https://github.com/The-OpenROAD-Project/OpenLane.git (fetch)
origin  https://github.com/The-OpenROAD-Project/OpenLane.git (push)
```

### To Reproduce

Clone the `openroad-bug` branch of https://github.com/MichaelBell/tt06-tinyQV recursively
Follow https://docs.google.com/document/d/1aUUZ1jthRpg4QURIIyzlOaPWlmQzr-jBn3wZipVUPt4/edit#heading=h.wwc5ldl01nl5 to set up the OpenLane flow for TinyTapeout, using the above project instead of the factory test project.
Run:

    tt/tt_tool.py --create-user-config
    tt/tt_tool.py --harden

Inspect `runs/wokwi/results/placement/tt_um_MichaelBell_tinyQV.nl.v` and `runs/wokwi/results/routing/tt_um_MichaelBell_tinyQV.nl.v` and note that the wires for the input for instance `_2354_` have been swapped:

```
 sky130_fd_sc_hd__and2b_1 _2354_ (.A_N(_0535_),
    .B(_0536_),
    .X(_0537_));
```
becomes
```
 sky130_fd_sc_hd__and2b_1 _2354_ (.A_N(_0536_),
    .B(_0535_),
    .X(_0537_));
```

QuantamHD, 2024-02-10:
@maliberty @tspyrou Please prioritize this issue!

maliberty, 2024-02-11:
Thanks for the report & fix.  I'll take a look and add a unit test.

closed 2024-02-12 by maliberty

maliberty, 2024-04-05:
@mattvenn pin swapping was introduced in #3215 and fixed here.  You would have to figure out what versions of OR was used in which tapeouts to know which could be affected.

In any tapeout you should be doing functional verification as no tool, at any price, can guarantee such problems can't happen.

commit cbb8ef34 (2024-02-12): rsz: prevent swapping of (!A) & B

There is more to improve here but at least all the swaps look legal.  Some
swaps are not detected but they are in more uncommon gates.

Fixes #4647