The-OpenROAD-Project/OpenROAD #4650: Buffer added immediately after input buffer

oharboe, 2024-02-10:
### Describe the bug

A buffer is added immediately after an input buffer for a signal that goes through (not registered) the macro below. It is as if the buffer after input is added without regards to the existing buffer.

To reproduce, run `make DESIGN_CONFIG=designs/asap7/mock-array/Element/config.mk` with https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/1794

To view the problem, run:

1. untar  [extra-buffer.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/14228657/extra-buffer.tar.gz)
2. Run `./run-me-mock-array_Element-asap7-base.sh`
3. Examine the timing path for endpoint `io_lsbOuts_6`
4. Observe the back-to-back buffers on the input

[image]

### Expected Behavior

Remove the superflous buffer

### Environment

```shell
OpenROAD v2.0-12143-g4aabf0d3c
```

### To Reproduce

See above

maliberty, 2024-02-12:
The buffer is kept from synthesis as it directly connects an input port to an output port (you can tell by the name _776_).  Later we buffer the input port and the output port so we get the three buffer chain.  I believe this has come up before.  The location of the buffer is just whatever global placement happens to pick (from an hpwl perspective the X coordinate has no effect).

We could do something like don't buffer the input if it only drives buffer(s) currently.  We would still buffer the output resulting in a two buffer net (prior to any repair).  Is that acceptable?

oharboe, 2024-02-12:
@maliberty That would be great, but it is a separate concern from being able to answer the question of what happens w zero buffers. I think both have value.

maliberty, 2024-02-12:
> zero

I don't understand what you comment about zero buffers relates to above.  Having zero buffers is a bad idea.

oharboe, 2024-02-12:
> > zero
> 
> I don't understand what you comment about zero buffers relates to above. Having zero buffers is a bad idea.

For the mock array, there is a discussion to be had as to how many buffers are needed. Maybe not for every Element. Therefore, it is interesting to do a measurement of what zero buffers would yield as a limit. If with zero buffers one can not.reach acceptable numbers, then that is a useful negative result for architectural feedback.

PR #4654 (2024-02-17): rsz: don't buffer input nets that are already connected to buffers
remove_buffers keeps a buffer between two top level ports.  We then add both input and output buffers which ends with a three buffer chain.  This reduces it to a two buffer net.

Fixes #4650
commit 91b57f02 (2024-02-17): rsz: don't buffer input nets that are already connected to buffers

remove_buffers keeps a buffer between two top level ports.  We then
add both input and output buffers which ends with a three buffer
chain.  This reduces it to a two buffer net.

Fixes #4650
closed 2024-02-17 by maliberty