The-OpenROAD-Project/OpenROAD #4652: Expalin "380.00  380.00   clock network delay (ideal)" with `report_checks -format full_clock_expanded`

oharboe, 2024-02-12:
### Description

1. untar [clock-explanation.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/14239429/clock-explanation.tar.gz)
2. Run `./run-me-mock-array-asap7-base.sh`
3. Enter `report_checks -from {io_ins_left_1[57]} -to {ces_1_7/io_ins_left[57]} -format full_clock_expanded`

How does the static timing analysis arrive at the `380.00  380.00   clock network delay (ideal)` line?

```
OpenROAD v2.0-12148-g1f2f38624 
[deleted]
>>> report_checks -from {io_ins_left_1[57]} -to {ces_1_7/io_ins_left[57]} -format full_clock_expanded
Startpoint: io_ins_left_1[57] (input port clocked by clock_vir)
Endpoint: ces_1_7 (rising edge-triggered flip-flop clocked by clock)
Path Group: clock
Path Type: max

  Delay    Time   Description
---------------------------------------------------------
   0.00    0.00   clock clock_vir (rise edge)
 380.00  380.00   clock network delay (ideal)
  45.00  425.00 ^ input external delay
   0.00  425.00 ^ io_ins_left_1[57] (in)
  11.59  436.59 ^ input629/Y (BUFx2_ASAP7_75t_R)
   0.17  436.76 ^ ces_1_7/io_ins_left[57] (Element)
         436.76   data arrival time

 180.00  180.00   clock clock (rise edge)
   0.00  180.00   clock source latency
   0.00  180.00 ^ clock (in)
  14.73  194.73 ^ wire2/Y (BUFx16f_ASAP7_75t_R)
  42.61  237.34 ^ wire1/Y (BUFx16f_ASAP7_75t_R)
  57.53  294.87 ^ clkbuf_0_clock/Y (BUFx24_ASAP7_75t_R)
  25.15  320.02 ^ clkbuf_1_0_0_clock/Y (BUFx24_ASAP7_75t_R)
  26.12  346.14 ^ clkbuf_2_1_0_clock/Y (BUFx24_ASAP7_75t_R)
  41.16  387.30 ^ clkbuf_3_2_0_clock/Y (BUFx24_ASAP7_75t_R)
  21.22  408.52 ^ ces_1_7/clock (Element)
 -10.00  398.52   clock uncertainty
   0.00  398.52   clock reconvergence pessimism
  34.72  433.24   library setup time
         433.24   data required time
---------------------------------------------------------
         433.24   data required time
        -436.76   data arrival time
---------------------------------------------------------
          -3.52   slack (VIOLATED)

```

### Suggested Solution

Split the mystery line to multiple lines so that it is more obvious how this number is calculated.

maliberty, 2024-02-12:
it comes from 
```
set_clock_latency     380 [get_clocks ${clk_name}_vir]
```

oharboe, 2024-02-12:
@maliberty There is a big difference in clock tree insertion latency after CTS and after final. Which latency should I use in .sdc?

If I use the insertion latency from final, much shorter, I get a ton of hold cells.

maliberty, 2024-02-12:
A ton of hold cells where?  On i/o paths?  Latency on a reg2reg path should be the same for start and end points.

Do you know why there is a large difference?

oharboe, 2024-02-12:
> A ton of hold cells where? On i/o paths? Latency on a reg2reg path should be the same for start and end points.
> 
> Do you know why there is a large difference?

Yes, on inputs. If I reduce clock latency, hold cells are added to compensate.

Is the clock latency specified manually in commw

closed 2024-02-12 by oharboe