The-OpenROAD-Project/OpenROAD #4666: Improve CTS for attached case

oharboe, 2024-02-14:
Which seems to have resulted in a strange CTS result, look at the macro leaf nodes:

[image]

Unbalanced at the leaf buffer:

[image]

_Originally posted by @oharboe in https://github.com/The-OpenROAD-Project/OpenROAD/discussions/4660#discussioncomment-8453920_

oharboe, 2024-02-14:
To view the result:

```
$ . vars-mock-array-asap7-base.sh
$ ODB_FILE=results/asap7/mock-array/base/4_1_cts.odb openroad -gui scripts/gui.tcl
```

The problem goes away if I revert https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/1799:

```
$ . vars-mock-array-asap7-base.sh
$ export CTS_ARGS="-insertion_delay -sink_clustering_enable -balance_levels -distance_between_buffers 60"
$ openroad -no_init ${SCRIPTS_DIR}/cts.tcl
```

Results in the expected CTS, 4 macros hanging off each buffer and not 4 macros + a buffer hanging off a buffer now seen on master:

[image]

oharboe, 2024-02-14:
@maliberty FYI, this seems to be the reason we see some regressions after the cleanup in https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/1799

maliberty, 2024-02-14:
I don't think there were any regression related to that PR.  mock array is unchanged by that PR.

oharboe, 2024-02-14:
> I don't think there were any regression related to that PR. mock array is unchanged by that PR.

Yes, but that is only because mock-array overrides CTS_ARGS completely. I only picked up the `-distance_between_buffers 60` because asap7 had it by default. My *intention* was to use the default arguments and only add `-insertion_delay`.

So... I can complete the cleanup of https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/1799 in https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/1802 when the feature request in this issue is completed.

maliberty, 2024-02-14:
If you mean we would have seen some regression with a different PR than was actually merged then I agree.

oharboe, 2024-02-14:
> If you mean we would have seen some regression with a different PR than was actually merged then I agree.

I think the sequencing of PRs and feature requests worked out well, even if it was not intended :-)

Otherwise everything would have stalled pending this feature request having been completed.

maliberty, 2025-08-30:
macro trees have had substantial improvements.  Reopen if still reproducible