The-OpenROAD-Project/OpenROAD #4690: YOSYS TCL interpreter returned error ?

Revenantx86, 2024-02-20:
### Describe the bug

When I try to synthesize my top module I get the following error. I tried to look for more detailed log but could not find anything. Any idea how I can fix it ?

Thank you very much for your time..

```
1. Executing Verilog-2005 frontend: ./designs/sky130hd/uart/src/uart_rx_top.v
2. Executing Verilog-2005 frontend: ./designs/sky130hd/uart/src/uart_rx.v
3. Executing Verilog-2005 frontend: ./designs/sky130hd/uart/src/fifo.v
4. Executing Liberty frontend: ./objects/sky130hd/uart_rx_module_top/base/lib/sky130_fd_sc_hd__tt_025C_1v80.lib
5. Executing Verilog-2005 frontend: /home/revenant/OpenROAD-flow-scripts/flow/platforms/sky130hd/cells_clkgate_hd.v
Using ABC speed script.
[FLOW] Extracting clock period from SDC file: ./results/sky130hd/uart_rx_module_top/base/clock_period.txt
[FLOW] Setting clock period to 5
6. Executing SYNTH pass.
6.1. Executing HIERARCHY pass (managing design hierarchy).
ERROR: TCL interpreter returned an error: Yosys command produced an error
Command exited with non-zero status 1
Elapsed time: 0:00.12[h:]min:sec. CPU time: user 0.11 sys 0.01 (98%). Peak memory: 42752KB.
``` 

### Expected Behavior

Normal synth flow

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 6.5.0-17-generic
os: Ubuntu 22.04.3 LTS (Jammy Jellyfish)
cmake version 3.24.2
-- The CXX compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-11974-gefb1f59c3
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
[... 124 lines cut ...]
# Verilog source files #
export VERILOG_FILES = \
					$(RTL_DIR)/uart_rx_top.v \
					$(RTL_DIR)/uart_rx.v \
					$(RTL_DIR)/fifo.v 

# Clock synth constraints #
export SDC_FILE      = $(TOP_DIR)/synthesis/rx_module/constraint.sdc

# Power #
export PDN_TCL = $(TOP_DIR)/script/macro_pdn.tcl

export ABC_DRIVER_CELL = sky130_fd_sc_hd__buf_1
export ABC_LOAD_IN_FF = 3

# Size #
export CORE_UTILIZATION = 75
export CORE_ASPECT_RATIO = 2
export PLACE_DENSITY ?= 0.90

# Routing
export MIN_ROUTING_LAYER = met1
export MAX_ROUTING_LAYER = met4 

# Met 5 Reserved for global routing

QuantamHD, 2024-02-22:
Since this appears to be a Yosys problem you might be better assisted by filing this bug in the Yosys project https://github.com/YosysHQ/yosys

maliberty, 2024-03-05:
I suggest you refile as suggested above.

closed 2024-03-05 by maliberty

tvd040562, 2024-04-11:
I have the same error.  The design ran fine in Openlane which also uses the same version of Yosys.

maliberty, 2024-04-11:
It could be the OL uses a different set of yosys commands that doesn't happen to trigger this issue.  It still remains a yosys issue not an OR one.

tvd040562, 2024-04-12:
@maliberty You are correct.  If I used the design name directly then it worked fine.  Passing the $::env(DESIGN_NAME) will cause the error.  I don't know how to workaround this problem yet, may be reinstalling the yosys tool separately from the ORFS.  Any idea?