The-OpenROAD-Project/OpenROAD #4703: HierRTLMP fail

diaoerlie, 2024-02-26:
### Describe the bug

hello, I've try run the RTLMP2 flow , but it just not work:
(nangate45/ariane136 + openroad flow)

1. datain: 
DESIGN_CONFIG=./designs/nangate45/ariane136/config.mk

2. how to run:
cd flow & make

3. what is error:
[WARNING STA-0354] set_input_delay relative to a clock defined on the same port/pin not allowed.
[INFO ORD-0030] Using 24 thread(s).
HierRTLMP Flow enabled...
rtl_macro_placer -max_num_inst 30000 -min_num_inst 5000 -max_num_macro 16 -min_num_macro 4 -halo_width 10 -report_directory ./objects/nangate45/ariane136/base/rtlmp
[ERROR STA-0402] rtl_macro_placer -max_num_inst is not a known keyword or flag.
Error: macro_place_util.tcl, 138 STA-0402
Command exited with non-zero status 1
Elapsed time: 0:01.31[h:]min:sec. CPU time: user 1.20 sys 0.11 (100%). Peak memory: 320756KB.
Makefile:616: recipe for target 'do-2_4_floorplan_macro' failed
make[1]: *** [do-2_4_floorplan_macro] Error 1
Makefile:616: recipe for target 'results/nangate45/ariane136/base/2_4_floorplan_macro.odb' failed

### Expected Behavior

RTLMP2 flow work and genarete the partition macro placememt result

### Environment

```shell
[WARNING PPL-0015] Macro i_cache_subsystem/i_nbdcache/valid_dirty_sram/genblk1_0__i_ram/macro_mem_1 is not placed.
[WARNING PPL-0015] Macro i_cache_subsystem/i_nbdcache/valid_dirty_sram/genblk1_0__i_ram/macro_mem_2 is not placed.
[WARNING PPL-0015] Macro i_cache_subsystem/i_nbdcache/valid_dirty_sram/genblk1_0__i_ram/macro_mem_3 is not placed.
Found 0 macro blocks.
Using 1u default distance from corners.
Using 2 tracks default min distance between IO pins.
[INFO PPL-0007] Random pin placement.
Elapsed time: 0:01.79[h:]min:sec. CPU time: user 1.59 sys 0.19 (99%). Peak memory: 330300KB.
bash: - : invalid option
Usage:	bash [GNU long option] [option] ...
	bash [GNU long option] [option] script-file ...
GNU long options:
	--debug
	--debugger
```

### To Reproduce

it's a close environment , and i cannot attarch file

maliberty, 2024-02-26:
Most likely you have an out of date version of OR.  Please include the commit ids for OR & ORFS.

diaoerlie, 2024-02-26:
@maliberty 
hello maliberty,

1. here's the ORFS git log:

/OpenROAD-flow-scripts_latest]$git rev-parse HEAD
97d54ba50c9e1213757524d9bd4e1a40d26c1a17

/OpenROAD-flow-scripts_latest]$git log | head -n 1 
commit 97d54ba50c9e1213757524d9bd4e1a40d26c1a17

2. here's the OR version:
/OpenROAD-flow-scripts_latest/tools/install/OpenROAD/bin]$./openroad -version
v2.0-4994-g0d653c681

closed 2024-02-27 by diaoerlie

maliberty, 2024-02-27:
I'm not sure what you mean by "PC server which is off-line".