The-OpenROAD-Project/OpenROAD #4708: OpenSTA (in ORFS) could not support mem.lib

diaoerlie, 2024-02-28:
### Describe the bug

I am trying to run ORFS by make do-floorplan, the flow is crash while reading mem.lib .

crash log
[WARNING STA-0125] /local2/OpenROAD-flow-scripts_latest/flow/platforms//lib/*.lib line 3225, table row has 7 columns but axis has 76.
[WARNING STA-0126] /local2/OpenROAD-flow-scripts_latest/flow/platforms//lib/*.lib line 3225, table has 7 rows but axis has 1.
Signal 11 received
Stack trace:
0# 0x0000000000E08223 in /local2/OpenROAD-flow-scripts_latest/tools/install/OpenROAD/bin/openroad
1# 0x00007F9E17108F10 in /lib/x86_64-linux-gnu/libc.so.6
2# sta::LibertyReader::makeTableAxis(int) in /local2/OpenROAD-flow-scripts_latest/tools/install/OpenROAD/bin/openroad
3# 0x000000000116A00D in /local2/OpenROAD-flow-scripts_latest/tools/install/OpenROAD/bin/openroad
4# sta::makeLibertyComplexAttr(char const, sta::Vectorsta::LibertyAttrValue*, int) in /local2/OpenROAD-flow-scripts_latest/tools/install/OpenROAD/bin/openroad
5# LibertyParse_parse() in /local2/OpenROAD-flow-scripts_latest/tools/install/OpenROAD/bin/openroad
6# sta::parseLibertyFile(char const, sta::LibertyGroupVisitor*, sta::Report*) in /local2/OpenROAD-flow-scripts_latest/tools/install/OpenROAD/bin/openroad
7# sta::readLibertyFile(char const*, bool, sta::Network*) in /local2/OpenROAD-flow-scripts_latest/tools/install/OpenROAD/bin/openroad
8# sta::Sta::readLibertyFile(char const*, sta::Corner*, sta::MinMaxAll const*, bool) in /local2/OpenROAD-flow-scripts_latest/tools/install/OpenROAD/bin/openroad
9# sta::Sta::readLiberty(char const*, sta::Corner*, sta::MinMaxAll const*, bool) in /local2/OpenROAD-flow-scripts_latest/tools/install/OpenROAD/bin/openroad
10# 0x0000000000FFFE70 in /local2/OpenROAD-flow-scripts_latest/tools/install/OpenROAD/bin/openroad
11# TclNRRunCallbacks in /tools/anaconda3/lib/libtcl8.6.so
12# 0x00007F9E1BCBA19F in /tools/anaconda3/lib/libtcl8.6.so
13# Tcl_EvalEx in /tools/anaconda3/lib/libtcl8.6.so
14# Tcl_Eval in /tools/anaconda3/lib/libtcl8.6.so
15# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in /local2/OpenROAD-flow-scripts_latest/tools/install/OpenROAD/bin/openroad
16# ord::tclAppInit(Tcl_Interp*) in /local2/OpenROAD-flow-scripts_latest/tools/install/OpenROAD/bin/openroad
17# Tcl_MainEx in /tools/anaconda3/lib/libtcl8.6.so
18# main in /local2/OpenROAD-flow-scripts_latest/tools/install/OpenROAD/bin/openroad
19# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
20# _start in /local2/OpenROAD-flow-scripts_latest/tools/install/OpenROAD/bin/openroad
Command terminated by signal 11

2. mem lib line 3225 is attached:
[image]

### Expected Behavior

could support this format mem.lib 
or by pass/ continue on error while readling lib

### Environment

```shell
fatal: unable to access 'https://github.com/The-OpenROAD-Project/OpenROAD.git/': Could not resolve host: github.com

the git log code is:
/local2/OpenROAD-flow-scripts_latest/tools/OpenROAD]$git log | head -n 1
commit 933ae804c19ee6216f3def610533cac9c99094f3
```

### To Reproduce

could not upload file

maliberty, 2024-02-28:
Can you provide the library or is it confidential?  CCS support should come in the near future but for now we only support NLDM so even if this parses you may not be able to use it yet.

diaoerlie, 2024-02-28:
hi maliberty , it's confidential, but i can provide some pattern of this file:

1. [WARNING STA-0125] line 2152, **table row has 29 columns but axis has 32.**

patten:
[image]

2. [WARNING STA-0130] lib line 3217, when attribute inside table model.
[image]

3. [WARNING STA-0126] line 3225, table has 7 rows but axis has 1
**Signal 11 received**
[image]

maliberty, 2024-02-28:
How was the .lib generated?

@tspyrou any ideas?

maliberty, 2024-02-28:
Can you give some info on the CELL_UP definition?

maliberty, 2024-02-28:
eta for ccs (not ccs-noise) is ~1 month for a preliminary version.

diaoerlie, 2024-03-01:
> info on the CELL_UP definition?

it's a mem rise delay, retaining_rise is mem wake up from retain mode

diaoerlie, 2024-03-05:
@maliberty is there any way to bypass this issue? because the timing condition is TEST mode

maliberty, 2024-03-05:
As I don't really know what the problem it is it hard to suggest something specific.  Can you use a .lib without the ccs data and just nldm?  Can you reduce the .lib to the barest minimum necessary to reproduce the issue in a non-confidential manner?

maliberty, 2024-03-16:
Please reopen if you can provide more information

closed 2024-03-16 by maliberty