The-OpenROAD-Project/OpenROAD #4710: OpenSTA: write_timing_model should add is_macro_cell true

maliberty, 2024-02-28:
### Description

write_timing_model should add to the generated .lib:
```
is_macro_cell : true;
```
The lack of this confuses
```
bool Resizer::isCombinational(LibertyCell* cell) const
{
  if (!cell) {
    return false;
  }
  return (!cell->isClockGate() && !cell->isPad() && !cell->isMacro()
          && !cell->hasSequentials());
}
```
into thinking a macro is combinational.

### Suggested Solution

Write the requested

maliberty, 2024-02-28:
asap7/aes-block demonstrates the issue.

maliberty, 2024-02-28:
Workaround in https://github.com/The-OpenROAD-Project/OpenROAD/pull/4711

tspyrou, 2024-03-01:
@maliberty update was pushed today.

closed 2024-03-12 by maliberty