The-OpenROAD-Project/OpenROAD #4719: Memory error in sta

maliberty, 2024-02-29:
### Describe the bug

With the latest update to sta in OR a memory bug has been discovered.  In ConcreteParasiticNetwork::ensureParasiticNode

```
    node = new ConcreteParasiticNode(pin, net != net_);
    pin_nodes_[pin] = node;
```

while in ConcreteParasiticNetwork::deleteNodes:
```
  for (auto pin_node : pin_nodes_) {
    ParasiticNode *node = pin_node.second;
    delete node;
  }
```

Since `class ParasiticNode {};` declares no virtual dtor this means the ~ConcreteParasiticNode is not called.  This code either needs to use a virtual dtor in the base class or at least use the dervied type in the loop (or auto).

### Expected Behavior

No memory errors

### Environment

```shell
OR de92b19f3ed9ec26e35d52602df935280d2f6041 
using
sta c367a000642e07b5be2bdf0f41275d304c3a5797
```

### To Reproduce

Proprietary design but should be easily reproduced on any other.

maliberty, 2024-02-29:
Workaround in https://github.com/The-OpenROAD-Project/OpenSTA/pull/228

tspyrou, 2024-03-01:
@maliberty This was fixed 3 weeks ago when the virtual methods on ConcreteParasiticNode were removed.
902a1bff8 2024-02-08 parasitics api update.

maliberty, 2024-03-01:
Ok I'll start updating again...

closed 2024-03-01 by maliberty