The-OpenROAD-Project/OpenROAD #4724: [Partitioner] Any documentation for multi-constraint example for TritonPart?

ombhilare999, 2024-03-01:
### Description

Hi,

I am using TritonPart in a personal project and my application closely resembled the following application mentioned in the TritonPart paper. [Here](https://vlsicad.ucsd.edu/Publications/Conferences/401/c401.pdf):

> In other words, the standard balance constraint must be satisfied
along each dimension of weights. A practical example of this
constraint is partitioning a netlist across multiple FPGAs, where
resources such as flip-flops (FFs), digital signal processing blocks
(DSPs), and look-up tables (LUTs) are all limited. This is a hard
constraint

1. In my application, I had three different types of vertices, denoted as t1, t2, and t3. My main concern was to ensure that no more than four crossings were allowed between any partitions. I believe to achieve this functionality; I need to set `-balance_constraint` to `3`.
2. In each individual partition or block, I could only have a maximum of `four t1 cells, 16 t2 cells, and four t3 cells`. This was the maximum condition. However, I was unsure about how to implement this constraint. I have read the OpenROAD paper and documentation, but any related reference would be very helpful for my application.

Thank you, let me know if you need any more information.

maliberty, 2024-03-11:
@bodhi91 can you help with this one too

bodhi91, 2024-03-11:
1. The -`balance_constraint` is always imposed on the total amount of vertex weights allowed on one partition. I assume by "crossings" you are referring to the cutsize of the partition. If this is the case, then setting the balance constraint to 3 will not work. The current version of TritonPart, unfortunately, does not allow constraints on cutsize (although I do realize in certain FPGA applications this is required; also something that we are working on currently). 
2. To implement this, you can leverage the -`base_balance` argument. If partition `X` can have a maximum of `a, b, c` vertices, I would set `base_balance` to `floor((a-y) / X), floor((b-y) / X), floor((c-y) / X)`, where `y` is small compared to `a, b, c`. Next, you can use a `balance_constraint` of `y/X`. To explain in more detail, you would want to set the `base_balance` in such a way that the `upper block balance` https://github.com/The-OpenROAD-Project/OpenROAD/blob/85f541bb67dd0498b0f53c78b572a12d0c4aa7a7/src/par/src/Hypergraph.cpp#L194  of any partition (calculated with the `balance_constaint`)  amounts to the maximum value of vertex weights you would want in your partition. 

Please try this and let me know if it works.

maliberty, 2024-03-11:
Close?

closed 2024-03-16 by maliberty