The-OpenROAD-Project/OpenROAD #4726: rcx test case not working

vijayank88, 2024-03-01:
### Describe the bug

Unable to run this test case: https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/src/rcx/test/generate_rules.tcl

Getting following error:

```
openroad generate_rules.tcl 
OpenROAD 762bba74f1f86b4626f0afa62c34a27c623eb2bb 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ODB-0222] Reading LEF file: sky130hs/sky130hs.tlef
[INFO ODB-0223]     Created 13 technology layers
[INFO ODB-0224]     Created 25 technology vias
[INFO ODB-0226] Finished LEF file:  sky130hs/sky130hs.tlef
[INFO ODB-0128] Design: blk
[INFO ODB-0130]     Created 141104 pins.
[INFO ODB-0133]     Created 70552 nets and 0 connections.
[INFO RCX-0001] Reading SPEF file: generate_pattern.spef
[ERROR ODB-0428] Cannot open file generate_pattern.spef for "r"
Error: generate_rules.tcl, 7 ODB-0428
```

### Expected Behavior

Add missing file path to validate the test case

### Environment

```shell
latest master
```

### To Reproduce

cd src/rcx/test
openroad generate_rules.tcl

### Relevant log output

```shell
[INFO RCX-0001] Reading SPEF file: generate_pattern.spef
[ERROR ODB-0428] Cannot open file generate_pattern.spef for "r"
Error: generate_rules.tcl, 7 ODB-0428
```

### Additional Context

Is there any instruction on how to run this steps: ` generate_pattern.spef needs to be generated from an OSS tool (eg magic)`

maliberty, 2024-03-02:
That is a disabled test that probably should just be removed.  It originally contained disallowed proprietary tool output that was removed.  Why are you trying to run it?

vijayank88, 2024-03-04:
I am trying to generate rcx.rules to use it for post layout timing report with ORFS with private PDK.

maliberty, 2024-03-04:
Ok but this test will not help you with that goal

closed 2024-03-04 by maliberty