The-OpenROAD-Project/OpenROAD #4751: CTS, automate selection of good ` -distance_between_buffers 60` value

oharboe, 2024-03-08:
### Description

Expected:

[image]

Actual:

[image]

To reproduce: untar  [cts.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/14541128/cts.tar.gz)

```
./run-me-mock-array-asap7-base.sh
. vars-mock-array-asap7-base.sh
ODB_FILE=results/asap7/mock-array/base/4_1_cts.odb openroad -gui scripts/gui.tcl
```

### Suggested Solution

CTS should automatically pick good defaults

oharboe, 2024-03-08:
> It seems that the timing repair step is degrading the clock skew. Some of timing repair can be pulled into CTS to mitigate this issue. @oharboe, what's the needed timeline for this enhancement?

I would prioritize orderly p

maliberty, 2024-03-08:
I would guess it is repair_clock_nets rather than repair_timing that is the problem (the later shouldn't touch the clock tree).

precisionmoon, 2024-03-08:
> I would guess it is repair_clock_nets rather than repair_timing that is the problem (the later shouldn't touch the clock tree).

Correct.