The-OpenROAD-Project/OpenROAD #4752: Boost checksum failure

gadfort, 2024-03-09:
### Describe the bug

The checksum seems to have changed to:
md5sum boost_1_80_0.tar.gz 
38631ccf5da80ab98a375a2a53052760  boost_1_80_0.tar.gz

### Expected Behavior

No checksum failure

### Environment

```shell
N/A
```

### To Reproduce

./etc/DependencyInstaller.sh

### Relevant log output

```shell
113.3 Saving to: 'boost_1_80_0.tar.gz'
113.3 
113.3      0K                                                       100%  141M=0s
113.3 
113.3 2024-03-09 03:29:57 (141 MB/s) - 'boost_1_80_0.tar.gz' saved [859/859]
113.3 
113.3 md5sum: WARNING: 1 computed checksum did NOT match
113.3 boost_1_80_0.tar.gz: FAILED
```

maliberty, 2024-03-09:
@vvbandeira any idea?

gadfort, 2024-03-09:
This appears to have been a result of sourceforge having some trouble yesterday. Appears to have been resolved.

closed 2024-03-09 by gadfort