The-OpenROAD-Project/OpenROAD #4757: alt+mousewheel no longer scrolls horizontally

oharboe, 2024-03-10:
### Describe the bug

1. open OpenROAD GUI
2. zoom in using ctrl+mousewheel
3. just mousewheel alone moves up and down
4. alt+mousewheel should have moved horizontally, but moves up/down 

OpenROAD v2.0-12475-g1b6a4e128

### Expected Behavior

alt+mousewheel should have moved horizontally, but moves up/down

### Environment

```shell
OpenROAD v2.0-12475-g1b6a4e128
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 23.10
Release:	23.10
Codename:	mantic
```

### To Reproduce

See above

maliberty, 2024-03-10:
There is no recent GUI change that I can imagine causing this.  Are you sure it isn't a local issue with your alt key?

oharboe, 2024-03-11:
Works on my workstation, but not on my laptop. Could this be because the laptop has a mousepad?

oharboe, 2024-03-17:
@maliberty I'm not able to get alt+mousewheel to work on my laptop, nor am I able to grab the view and move it around with the mouse. What was the key to grab the view and move it again?

Could someone try on an Ubuntu laptop to confirm?

maliberty, 2024-03-17:
> What was the key to grab the view and move it again?

middle mouse button click and drag

oharboe, 2024-03-17:
> > What was the key to grab the view and move it again?
> 
> middle mouse button click and drag

I dont have a middle button

maliberty, 2024-03-17:
Usually there is some way to emulate it on systems that don't have it - some combinations of key press plus button.

maliberty, 2024-03-17:
https://askubuntu.com/questions/1422992/middle-click-not-emulated-in-22-04

oharboe, 2024-03-18:
> https://askubuntu.com/questions/1422992/middle-click-not-emulated-in-22-04

Holding down mousewheel to grab works on my workstation.

maliberty, 2024-03-18:
I don't know that we have anyone with an Ubuntu laptop.  Are you stuck?

maliberty, 2024-03-28:
@oharboe are you stuck or should we close?

oharboe, 2024-04-02:
Let's have a look next time we meet, might be trivial.

oharboe, 2024-04-17:
hold two fingers horizontally and vertically, respectively and move vertical/horizontally

closed 2024-04-17 by oharboe