The-OpenROAD-Project/OpenROAD #4762: Add skew column to timing report

oharboe, 2024-03-11:
### Description

It would be nice to see a clock tree skew column to immediately see get a sense for which timing paths that have a problem with a timing skew vs. problem with the path itself:

[image]

[image]

### Suggested Solution

Add skew column

tspyrou, 2024-03-27:
Pls make class clkskew in sta/search/ClkSkew.cc public to allow the GUI access.

tspyrou, 2024-04-02:
@AcKoucher Cherry will make a proposal on a new API. We discussed it today.

AcKoucher, 2024-04-02:
@tspyrou Ok!

oharboe, 2024-04-03:
@tspyrou How can the userinterface indicate if the skew is useful or not for a path?

Could there be a "Useful skew" column where positive is useful and negative is unhelpful skew?

jjcherry56, 2024-04-04:
Making class ClkKkew (not clkskew) in sta/search/ClkSkew.cc public would not do you any good because there is no api to populate it for a given source/target pair in the timing summary report.
I added PathEnd::clkSkew which is one-to-one with the reported lines in commit
1d7bd5ba PathEndCheck::clkSkew

maliberty, 2024-04-04:
This API has been integrated to OR with #4906

PR #4911 (2024-04-06): gui: add skew to timing report
Resolve #4762
closed 2024-04-06 by maliberty