The-OpenROAD-Project/OpenROAD #4763: [WARNING DRT-0422] No routing tracks pass through the center of Term io_outs_up[28]

oharboe, 2024-03-11:
### Describe the bug

Based on `make DESIGN_CONFIG=designs/asap7/mock-array/Element/config.mk`

Untar [detailed-route-warning.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/14558177/detailed-route-warning.tar.gz)

```
$ NUM_CORES=16 ./run-me-mock-array_Element-asap7-base.sh
OpenROAD v2.0-12488-g85f541bb6
 [deleted]
[WARNING DRT-0422] No routing tracks pass through the center of Term clock
[WARNING DRT-0422] No routing tracks pass through the center of Term io_ins_down[0]
[WARNING DRT-0422] No routing tracks pass through the center of Term io_ins_down[10]
[WARNING DRT-0422] No routing tracks pass through the center of Term io_ins_down[11]
[WARNING DRT-0422] No routing tracks pass through the center of Term io_ins_down[12]
[WARNING DRT-0422] No routing tracks pass through the center of Term io_ins_down[13]
[WARNING DRT-0422] No routing tracks pass through the center of Term io_ins_down[14]
[WARNING DRT-0422] No routing tracks pass through the center of Term io_ins_down[15]
[deleted]
```

### Expected Behavior

No warning and no problems

### Environment

```shell
OpenROAD v2.0-12488-g85f541bb6
```

### To Reproduce

See above

maliberty, 2024-03-11:
It looks like the routing track is not through the center as the message indicates.  How was the pin placed?

maliberty, 2024-03-11:
It would be helpful to turn off all other layers in the image.

oharboe, 2024-03-11:
> It looks like the routing track is not through the center as the message indicates. How was the pin placed?

standard orfa flow, no changes.

eder-matheus, 2024-03-11:
The pins are aligned with the tracks:
[image]

I've checked it with the ruler, and the pin placement looks ok. This is probably a false warning from DRT, which I'll investigate this week.

closed 2024-03-16 by maliberty