The-OpenROAD-Project/OpenROAD #4769: PDN generation fails to create VIAs to power macro if there are no margins

smunaut, 2024-03-11:
### Describe the bug

If you have no margin in your design (so core area == die area), then the PDN straps will go from one end of the die to the other.

In `GridComponent::addShape` ( `OpenROAD/src/pdn/src/grid_component.cpp` line 158 ), we find :

```c
  if (final_shape_rect.xMin() == die_area.xMin()) {
    const int x = std::min(static_cast<int>(die_area.xMin() + min_width),
                           final_shape_rect.xMax());
    odb::Rect pin_rect = final_shape_rect;
    pin_rect.set_xhi(x);
    shape_ptr->addBTermConnection(pin_rect);
  }
```

So those shapes end up with BTerm attached to them.

When comes the time to create the via going from the PDN strips down to macro pads, in `Connect::makeVia` ( `OpenROAD/src/pdn/src/connect.cpp` line 506 ) :

```c
      if (upper->getLayer() == l1) {
        if (!upper->isModifiable() || upper->hasTermConnections()) {
          // upper is not modifiable to all sides must fit
          skip_caching = true;
          upper_constraint.must_fit_x = true;
          upper_constraint.must_fit_y = true;
          upper_constraint.intersection_only = false;
        } else {
          upper_constraint.must_fit_x = !upper->isHorizontal();
          upper_constraint.must_fit_y = !upper->isVertical();
        }
      }
```

So this changes the constraints and then when it actually tests those, it fails because `must_fit_x` fails the check, even though it completely fits in the X direction, it just limits the comparaison to the overlap area rather than the complete existing shape ...

### Expected Behavior

I expect the vias to be created

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 5.10.0-19-cloud-amd64
os: Debian GNU/Linux 11 (bullseye)
cmake version 3.18.4
-- The CXX compiler identification is GNU 10.2.1
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-12488-g85f541bb6
-- System name: Linux
-- Compiler: GNU 10.2.1
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 10.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
[... 126 lines cut ...]
-- Check if compiler accepts -pthread - yes
-- SSTA: 0
-- Found SWIG: /usr/bin/swig4.0 (found suitable version "4.0.2", minimum required is "3.0")  
-- STA executable: /mnt/pdk/OL2/native/OpenROAD/src/sta/app/sta
-- Configuring incomplete, errors occurred!
```

### To Reproduce

There is a reproduce case attached (call `run.sh`).
[via-issue.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/14564271/via-issue.zip)

### Relevant log output

```shell
[WARNING PDN-0110] No via inserted between met4 and met5 at (99.3600, 131.3200) - (100.5600, 137.5200) on vssd1
[WARNING PDN-0110] No via inserted between met4 and met5 at (99.3600, 188.4400) - (100.5600, 194.6400) on vssd1
[WARNING PDN-0110] No via inserted between met4 and met5 at (107.3600, 131.3200) - (108.5600, 137.5200) on vssd1
[WARNING PDN-0110] No via inserted between met4 and met5 at (107.3600, 188.4400) - (108.5600, 194.6400) on vssd1
```

### Screenshots

How it is : https://i.imgur.com/ASchSLH.png
How it should be : https://i.imgur.com/cQiy2i0.png

maliberty, 2024-03-11:
@gadfort I've discussed this with Sylvain but reached the point where I could use your insight.

smunaut, 2024-03-11:
https://open-source-silicon.slack.com/archives/C0161A4A59V/p1710173064135569

gadfort, 2024-03-14:
@smunaut thanks for the detailed issue, made it pretty easy to fix.

PR #4789 (2024-03-14): pdn: fix behavior around bterms and vias
Fixes: #4769
closed 2024-03-14 by maliberty