The-OpenROAD-Project/OpenROAD #4781: Make GPL routability mode default on GPL.

maliberty, 2024-03-12:


gudeh, 2024-03-14:
Related PRs: #4658 for OR and https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/1840 for ORFS

gudeh, 2024-04-08:
Merged. PRs: [#4658](https://github.com/The-OpenROAD-Project/OpenROAD/pull/4658) and [#1915](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/1915).

Issues discovered: 
- GPL bug (same as happened with Cho): when accessing grid tiles from global cells. Solved.
- False divergence: solved by making the divergence check less restrictive.
 - Never ending DRC during DRT. It stopped occurring after merge with master. 
- MPL2 issue with Microwatt.
- Antenna check. Open issue. Work around by using previous target RC on one design.

closed 2024-04-08 by gudeh