The-OpenROAD-Project/OpenROAD #4786: GRT: update congestion estimation in GCells when fat vias are necessary

eder-matheus, 2024-03-13:


rovinski, 2024-03-14:
Is this a 50% reduction to the layer or a 50% reduction to layer transitions? It seems overly conservative to reduce the layer by 50%.

maliberty, 2024-03-14:
There is no via capacity modeling in fastroute so it in in layer resource.  The problem is that with only a fat via you block any adjacent track if wrong way is not allowed whenever you use a via.  We see this as a source of hard to fix problems in a particular PDK.  Once we can support the full set of wrong way rules this may be less needed and can be revisited.  50% is a rough but conservative guess.  It typically only occurs over macros so the general impact is limited.

rovinski, 2024-03-14:
I see, so this is reduction is intended only for subset of GCells and not all GCells? Makes sense.

closed 2024-03-21 by maliberty