The-OpenROAD-Project/OpenROAD #4792: Hierarchy browser shows flattened design

scorbetta, 2024-03-14:
### Describe the bug

After a successful run of the OpenLane flow, I am not able to see the hierarchical design from the Hierarchy Browser. It seems everything is flattened, and hence I am not able to highlight portions of the design.

### Expected Behavior

Hierarchy access

### OpenROAD Environment

```shell
There's no such `Env.sh` script.
```

### OpenLane Environment

```shell
open_pdks cd1748bb197f9b7af62a54507de6624e30363943
WARNING: issue-survey appears to be running inside the OpenLane
container.

This makes it difficult to rule out issues with your
environment.

Unless instructed specifically to do so, please run this command
outside the OpenLane container.
---

Kernel: Linux v6.5.0-25-generic
Distribution: centos 7
Python: v3.6.8 (OK)
OpenLane Git Version: 9dbd8b5ea2bd891bed4dcc97df5c7439083f0368
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

9dbd8b5e 2024-01-11T15:26:56+02:00 Fix `-synth_explore` crash (#2085) - Kareem Farid -  (HEAD, tag: 2024.01.12, origin/master, origin/HEAD, master)
a005df1f 2024-01-08T10:51:06+02:00 Updated link to newer version of spanish docs (#2082) - Laboratorio de Investigación en Microelectrónica y Arquitectura de Computadoras, EIE -- UCR -  (tag: 2024.01.09)
8bea8c51 2023-12-26T13:17:21+02:00 Update `OpenROAD` (#2062) - Kareem Farid -  (tag: 2023.12.27)
---
Git Remotes

origin	git@github.com:The-OpenROAD-Project/OpenLane.git (fetch)
origin	git@github.com:The-OpenROAD-Project/OpenLane.git (push)
```

### To Reproduce

The following is what I issued once the `HL_NEURON` design folder is ready

```
./flow.tcl -design HL_NEURON -tag run_flat -overwrite
```

Archive of the sources, configuration file and run folder can be downloaded from here: [https://we.tl/t-gA8SYCmXMH](url)

### Screenshots

file:///home/scorbetta/Pictures/Screenshots/Screenshot%20from%202024-03-14%2014-26-34.png

### Additional Context

Previous discussion: [https://github.com/The-OpenROAD-Project/OpenROAD/discussions/4742](url)

maliberty, 2024-03-15:
> There's no such `Env.sh` script.

https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/etc/Env.sh

The provided url doesn't work.  Most likely you have flattened the design in synthesis.  Please check if the incoming Verilog to OR has any module hierarchy.

scorbetta, 2024-03-15:
There are two Verilog files out of synthesis, one with hierachy named `HL_NEURON.hierarchy.nl.v` and one flattened named `HL_NEURON.v`. Which one does OR use?

scorbetta, 2024-03-15:
Also, the config file contains

```
set ::env(SYNTH_NO_FLAT) 1
set ::env(SYNTH_FLAT_TOP) 0
```

scorbetta, 2024-03-15:
https://we.tl/t-gA8SYCmXMH

vijayank88, 2024-03-15:
as per description in OpenLane: https://openlane.readthedocs.io/en/latest/reference/configuration.html
```
A flag that disables flattening the hierarchy during synthesis, only flattening it after synthesis, mapping and optimizations.
Enabled = 1, Disabled = 0 
```
End of synthesis stage flow will generate only flattened netlist.

scorbetta, 2024-03-15:
@vijayank88 Are you saying that the post-synthesis flow works only on flattened netlist irrespective of the configuration?

maliberty, 2024-03-15:
OR works with a flat or hierarchical Verilog.  Which one you load is up to your flow and isn't an OR issue per se.

maliberty, 2024-03-16:
we.tl does not work - it seems to require an account.  There is no actionable test case here.

scorbetta, 2024-03-18:
Can you please check the link again? I've tried from a couple of computers and it works. I just had to click "I agree".
Thanks

Here's the URL: https://we.tl/t-gA8SYCmXMH

maliberty, 2024-03-18:
I managed to download it but it is a just a bunch of files and not a reproducible test case (there is no script to invoke).  This is not in a form where any action can be taken.  

./runs/run_flat/results/synthesis/HL_NEURON.v contains no hierarchy.   I still think you have a problem upstream in synthesis not OR.

maliberty, 2025-08-28:
stale

closed 2025-08-28 by maliberty