The-OpenROAD-Project/OpenROAD #4809: floorplan fails to ready synthsized verilog

oharboe, 2024-03-18:
### Describe the bug

untar https://drive.google.com/file/d/1bDdfO-mr-abcCHUAktWgKzGhbZK6KqpS/view?usp=sharing

```
$ ./run-me-BoomTile-asap7-base.sh 
OpenROAD v2.0-12608-g50753a4c5 
[deleted]
[ERROR STA-0164] bazel-out/k8-fastbuild/bin/results/asap7/BoomTile/base/1_synth.v line 6228427, syntax error
Error: floorplan.tcl, 3 STA-0164
openroad> 
```

Line 6228427:

```
    .ARGS_WIDTH(32'b00000000000000000000000000000000),
```

Bigger excerpt:

```
  \$check  #(
    .ARGS_WIDTH(32'b00000000000000000000000000000000),
    .FLAVOR("assert"),
    .FORMAT(),
    .PRIORITY(32'b11111111111111111111111111111010),
    .TRG_ENABLE(32'b00000000000000000000000000000001),
    .TRG_POLARITY(1'b1),
    .TRG_WIDTH(32'b00000000000000000000000000000001)
  ) assert__assert_101 (
    .A(casez_tmp_539),
    .ARGS(),
    .EN(_GEN_1558),
    .TRG(clock)
  );
```

### Expected Behavior

Parse without problems.

### Environment

```shell
OpenROAD v2.0-12608-g50753a4c5
```

### To Reproduce

See above

oharboe, 2024-03-18:
I extracted the Rob.sv file into a standalone ORFS test: https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/1852

oharboe, 2024-03-18:
Small reproduction case:

untar [rob-floorplan-error.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/14637805/rob-floorplan-error.tar.gz)

```
$ ./run-me-rob-asap7-base.sh 
OpenROAD v2.0-12608-g50753a4c5 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ODB-0222] Reading LEF file: /home/oyvind/OpenROAD-flow-scripts/flow/foo/platforms/asap7/lef/asap7_tech_1x_201209.lef
[INFO ODB-0223]     Created 24 technology layers
[INFO ODB-0224]     Created 9 technology vias
[INFO ODB-0226] Finished LEF file:  /home/oyvind/OpenROAD-flow-scripts/flow/foo/platforms/asap7/lef/asap7_tech_1x_201209.lef
[INFO ODB-0222] Reading LEF file: /home/oyvind/OpenROAD-flow-scripts/flow/foo/platforms/asap7/lef/asap7sc7p5t_28_R_1x_220121a.lef
[INFO ODB-0225]     Created 212 library cells
[INFO ODB-0226] Finished LEF file:  /home/oyvind/OpenROAD-flow-scripts/flow/foo/platforms/asap7/lef/asap7sc7p5t_28_R_1x_220121a.lef
[ERROR STA-0164] ./results/asap7/rob/base/1_synth.v line 316224, syntax error
Error: floorplan.tcl, 3 STA-0164
openroad> 
```

maliberty, 2024-03-18:
I don't understand what this netlist is supposed to be.  It doesn't seem to be a gate-level netlist.  Is \$check a cell of some sort?  The name suggests some sort of assert.  It isn't present in 1_synth.v.  This seems more like a yosys issue.

oharboe, 2024-03-18:
@maliberty So yosys versioning issue? It used to work, but doesn't anymore?

maliberty, 2024-03-18:
I don't know what happened in old versions of yosys as this isn't part of our test suite.  You could roll back and see but it looks like you found a solution in your ORFS PR.

oharboe, 2024-03-18:
> I don't know what happened in old versions of yosys as this isn't part of our test suite. You could roll back and see but it looks like you found a solution in your ORFS PR.

0.38 failed in some other way. I don't have a lot of curiosity about this if we have a solution. I *suspect* that yosys grew a feature and that the output for yosys can now potentially be indigestible to openroad without https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/1853

QuantamHD, 2024-03-19:
These are internal Yosys nodes that are not removed by default when emitting verilog.

oharboe, 2024-03-19:
> These are internal Yosys nodes that are not removed by default when emitting verilog.

Fixed in master

closed 2024-03-19 by oharboe

ashkanr65, 2024-11-13:
> > These are internal Yosys nodes that are not removed by default when emitting verilog.
> 
> Fixed in master

On yosys 0.47+46 (git sha1 9da734100, g++ 11.4.0-1ubuntu1~22.04 -fPIC -O3) I have the same issue.