The-OpenROAD-Project/OpenROAD #4812: add "-Werror" option to openroad, warnings are errors

oharboe, 2024-03-19:
### Description

It would be nice to enable warnings are errors on testcases in ORFS once they have no errors.

### Suggested Solution

Implement a "-Werror" option. Naming idiom taken from gcc options, so something more openroadesque in terms of naming for option would be a good idea, perhaps `-error_on_warning`?

maliberty, 2024-03-19:
This already exists https://github.com/The-OpenROAD-Project/OpenROAD/blob/594c45876789352b0aad21b1edbc5892f1a4c070/src/CMakeLists.txt#L180

maliberty, 2024-03-19:
It is set to OFF in the CI but we don't want to have users get stuck on building with various compiler versions.

oharboe, 2024-03-19:
I dont mean when compiling openroad, but when running openroad.

maliberty, 2024-03-19:
We could add the option though I'm not sure if we can commit to zero warnings all the time.

maliberty, 2024-03-19:
I would make it a command in OR not a command line option.

oharboe, 2024-03-19:
Hmmm.... perhaps start w a CI perspective? Parse output and add warnings to baseline data(metadata), which would make this specific feature request obsolete.

oharboe, 2024-03-20:
https://github.com/The-OpenROAD-Project/OpenROAD/pull/4817 is much nicer.

closed 2024-03-20 by oharboe