The-OpenROAD-Project/OpenROAD #4820: Add clock period information to histogram

oharboe, 2024-03-20:
### Description

I found myself not knowing the clock period when I was explaining the histogram.

[image]

### Suggested Solution

Some sort of indication of clock period, perhaps, "Slack [ps] clock period 8500"?

maliberty, 2024-03-20:
You could have multiple clocks with differing periods.  Would a list be useful?

oharboe, 2024-03-20:
> You could have multiple clocks with differing periods. Would a list be useful?

Ah. Forgot about that. I guess a list then, yes.

maliberty, 2024-03-20:
Probably - are you volunteering ?

gadfort, 2024-03-20:
@maliberty not at this time, just a suggestion :-)

maliberty, 2024-03-20:
In that case I would avoid expanding the scope of the request.